repetier/Repetier-Host #327: Repetier host has stopped working

dharmanos, 2015-01-01:
Can't run repetier host on Win 7 with Nvidia geforce 315.
get EventType=CLR20r3 in event viewer

I uninstalled and reinstalled several time with deleting registry folders in between.
using .Net 4.5.1 
uninstalled and reinstalled .net a few times too.
registry values don't seem to be missing any data.

repetier, 2015-01-02:
What does the error say in the details?

dharmanos, 2015-01-02:
There are no details. I get a repterier host has stopped working box and it says windows is checking for a solution to the problem with a cancel button. Then the box fades away.

dharmanos, 2015-01-02:
Here are the details from the application log in event viewer:

Fault bucket , type 0
Event Name: CLR20r3
Response: Not available
Cab Id: 0

Problem signature:
P1: repetierhost.exe
P2: 1.0.0.0
P3: 515ef1c4
P4: System.Drawing
P5: 2.0.0.0
P6: 530f04ea
P7: 7b0
P8: 70
P9: System.ArgumentException
P10: 

Attached files:
C:\Users\Dave\AppData\Local\temp\WER92A5.tmp.WERInternalMetadata.xml

These files may be available here:
C:\Users\Dave\AppData\Local\Microsoft\Windows\WER\ReportArchive\AppCrash_repetierhost.exe_7af692af39d1f761f1513b77d14461eb56b025a4_110ca931

Analysis symbol: 
Rechecking for solution: 0
Report Id: d908eb36-928c-11e4-ab53-6c626d9f6b81
Report Status: 1

repetier, 2015-01-02:
That only shows it crashed inside a windows .NET module expecting different arguments.

Since you said it stopped working I assume it worked before. What did you change so that it does not work any more? Anything updated/installed? Also you should know it uses latest .net 4.0 not 4.5 also older versions are included in 4.5 as fas as I know. Only not sure what the latest version is.

dharmanos, 2015-01-02:
no. it never worked.  Should I rollback to .net 4.0?

repetier, 2015-01-02:
After some googling I found that 4.5.2 replaces 4.0 and should be 100% backward compatible. My developing machine also has 4.5.2 so that should not really be the reason.

I assume your card supports OpenGL 1.5 or higher? Normally NVidea does. The bad thing is that I can not find the reason remotely. Have you tried an older version like 0.95F? Befor you downgrade you must uninstall including registry when the question comes!

repetier, 2015-01-02:
One more thing - are you running windows in true color mode (24 or 32 bit)

dharmanos, 2015-01-02:
i'm using the solidoodle version which is 0.85 i think.
https://s3.amazonaws.com/SolidoodleFirmware/setupRepetierHostSolidoodle_0_85_117.exe

I also tried to install Repetier-Host 1.06 from Repetier.com but got the same results

This video card supports openGL 3.2.

I am running true color 32 bit.

I really appreciate your help :-)

dharmanos, 2015-01-02:
rolling back to .net 4.0 didn't help.

dharmanos, 2015-01-04:
Is there a way to find out what causes the program to stop? Like a script tracing program?

repetier, 2015-01-05:
You could build 0.90D from the sources in this repository and run them in VisualStudio. That should give you a exact traceback and you could most proably trace it down to a call from the host.

dharmanos, 2015-01-05:
Thanks. Visual studio reported:

An unhandled exception of type 'System.ArgumentException' occurred in System.Drawing.dll

Additional information: Font 'Arial' does not support style 'Regular'.

I installed it via this update: 
http://www.microsoft.com/en-us/download/confirmation.aspx?id=16083

Now everything works.

repetier, 2015-01-05:
How should someone think of that :-) Great you found it. Will put it in FAQ.