repetier/Repetier-Host #328: Repetier host crash when switching to preview tab

nzbart, 2015-01-04:
After slicing an object, Repetier will crash whenever I return to the Preview tab.

##Steps to reproduce on my machine##

1. Launch Repetier host 1.0.6.
2. Load and slice any object.
3. Click on Manual Control tab.
4. Click on Preview tab. Repetier host will crash with "RepetierHost has stopped working".

##Additional information##

I am running on a fully patched Windows 8.1 installation (6.3.9600 N/A Build 9600).

Start-up log:
```
13:39:06.720 : OpenGL version:1.1.0
13:39:06.735 : OpenGL extensions:GL_WIN_swap_hint GL_EXT_bgra GL_EXT_paletted_texture
13:39:06.735 : OpenGL renderer:GDI Generic
13:39:06.735 : Fast VBOs for rendering not supported. Using slower default method.
```

Information from event log:
```
Application: RepetierHost.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AccessViolationException
Stack:
   at RepetierHost.model.GCodePath.RefillVBO()
   at RepetierHost.model.GCodeVisual.BufferVBOs()
   at RepetierHostExtender.geom.ThreeDView.BufferVBOs()
   at RepetierHost.view.ThreeDControl.SetView(RepetierHostExtender.geom.ThreeDView)
   at RepetierHost.Main.assign3DView()
   at System.EventHandler.Invoke(System.Object, System.EventArgs)
   at System.Windows.Forms.TabControl.OnSelectedIndexChanged(System.EventArgs)
   at System.Windows.Forms.TabControl.WmSelChange()
   at System.Windows.Forms.TabControl.WndProc(System.Windows.Forms.Message ByRef)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.UnsafeNativeMethods.SendMessage(System.Runtime.InteropServices.HandleRef, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.UnsafeNativeMethods.SendMessage(System.Runtime.InteropServices.HandleRef, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.Control.SendMessage(Int32, IntPtr, IntPtr)
   at System.Windows.Forms.Control.WmNotify(System.Windows.Forms.Message ByRef)
   at System.Windows.Forms.Control.WndProc(System.Windows.Forms.Message ByRef)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.UnsafeNativeMethods.CallWindowProc(IntPtr, IntPtr, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.UnsafeNativeMethods.CallWindowProc(IntPtr, IntPtr, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.NativeWindow.DefWndProc(System.Windows.Forms.Message ByRef)
   at System.Windows.Forms.Control.WmMouseDown(System.Windows.Forms.Message ByRef, System.Windows.Forms.MouseButtons, Int32)
   at System.Windows.Forms.Control.WndProc(System.Windows.Forms.Message ByRef)
   at System.Windows.Forms.TabControl.WndProc(System.Windows.Forms.Message ByRef)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr, Int32, IntPtr, IntPtr)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG ByRef)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG ByRef)
   at System.Windows.Forms.Application+ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr, Int32, Int32)
   at System.Windows.Forms.Application+ThreadContext.RunMessageLoopInner(Int32, System.Windows.Forms.ApplicationContext)
   at System.Windows.Forms.Application+ThreadContext.RunMessageLoop(Int32, System.Windows.Forms.ApplicationContext)
   at RepetierHost.Program.Main(System.String[])
```

repetier, 2015-01-04:
Your problem seems to be

13:39:06.720 : OpenGL version:1.1.0
13:39:06.735 : OpenGL extensions:GL_WIN_swap_hint GL_EXT_bgra GL_EXT_paletted_texture
13:39:06.735 : OpenGL renderer:GDI Generic

which means your graphic card has no OpenGL and it uses a bad windows fallback that does not support VBOs. Nonetheless you seem to use VBOs. 

Installing the OpenGL driver for your card should help and make it even 1000 times faster.

nzbart, 2015-01-04:
Unfortunately, I have the latest Intel drivers for Windows 8.1. It looks like I'll either need to:
1. Reformat with Linux , which apparently supports OpenGL 2.0 for this card.
2. Properly disable VBOs.

Do you know of any way to disable the use of VBOs in Repetier so that I can avoid spending the time it will take to install & configure Linux on this machine?

repetier, 2015-01-04:
You can disable it in 3d settings. Select drawing method arrays as a solution, but do not enable filament visualization while printing. That will be too slow.

What card do you have that it has no OpenGL driver. Even Xp had OpenGL 1.5 already. Use the driver from graphic card vendor, not the windows driver! That can make the difference.

jon-bondy, 2015-01-04:
When MicroSoft forced everyone to upgrade to Win 8.1 about 2 months ago, 
it forced many of us to be unable to use our current drivers, and AMD 
has stopped supporting a lot of cards.  I am now in MUCH worse shape, 
because of MicroSoft, than I was with Win 8.0.

On 1/4/2015 10:22 AM, repetier wrote:
> You can disable it in 3d settings. Select drawing method arrays as a
> solution, but do not enable filament visualization while printing.
> That will be too slow.
>
> What card do you have that it has no [...]

nzbart, 2015-01-08:
I tried switching to the "Arrays" Draw Method, but it still crashes in the same way.

Interestingly, after I slice an object, it is correctly rendered in the Preview tab. It is only when I switch to another tab and then back to the Preview tab that Repetier crashes.

I use on-board graphics. Windows identifies it as: "Intel(R) G33/G31 Express Chipset Family (Microsoft Corporation - WDDM 1.0)". Intel doesn't seem to have a more up-to-date driver.

szlovak, 2016-05-15:
I have similar problem. I'm using Intel HD graphics 3000 on windows 10 64bit. Only available drivers are these from windows, no any drivers on intel's website. Everything is fine until I switched to shell thickness of 0.1mm and top/bottom shell thickness of 0.1mm also. It was really simple object, some raised text on thin rectangular. I can manually step each layer until last layer. If I choose to show complete object's preview it crashes immediately, only as soon filament is set to visible. I use RH 1.6.1 

But for sure it's not the only reason for crashing. It crashes when more complex objects are previewed, with regular settings and when bottom or top layer is previewed. I can omit those and I can preview rest of the layers.

repetier, 2018-11-02:
In that case you should at least use repetier-server to print even if you print using host. Host has a repetier-server ocnnector. That way a host crash will not stop print.

I wonder that slicer makes a difference if it crashes or not.  Both use same functions to visualize.

In any case should you learn how to make host use the OpenGL driver. This is often a problem with dual graphic cards - one for slow and one for fast processing. Try giving graphics 100% power in settings that often causes the switch to the good graphic card with OpenGL Support.