repetier/Repetier-Host #330: Idea - Extruded Filament Feedback Loop Using Optical Sensor

rf1k-mjh11, 2015-03-03:
Unfortunately I'm not a programmer or good with electronics, so I'm hoping someone here agrees with the feasibility of my idea and works on implementing it in the host software.

I recently had the idea of using the optical sensor that can be found in a generic PC mouse to track filament use.  The sensor could track the filament directly, or better the idler/pressure wheel (due to concerns with colors and transparent filament).  These generic PC mice cost less than $10 (€10) and boast a resolution of 1000dpi (0.025mm).  That should be plenty for the task at hand.

The signal (processed already in the mouse itself?) is sent to the PC wirelessly or via cable to a USB port, where the host software can access it.  Since the host knows exactly which GCodes have been sent, and how much filament should have been extruded, it can compare the received signal and react accordingly to events such as end of filament, clogged extruder, etc.

There may even be the possibility of a temperature-control feedback loop.  Since increased pressure in the extruder can lead to increased slip at the drive wheel, the host could raise the temperature slightly, thereby reducing the pressure.

I hope someone thinks the idea is worth pursuing,

mjh11

nophead, 2015-03-03:
A simpler solution is to use a proper hobbed drive pulley so the feed rate doesn't vary until you reach the limit where it slips.

rf1k-mjh11, 2015-03-03:
Even a properly hobbed drive pulley doesn't protect you from running out of filament, or the stepper motor skipping steps - but the optical sensor setup I'm proposing would.

And even without the sensor, QUOTE:

A simpler solution is to use a proper hobbed drive pulley so the feed rate doesn't vary until you reach the limit where it slips.

UNQUOTE
Okay, so, for whatever reason, the limit has been reached and the filament starts to slip - what now?  You're asleep, at dinner or at work - too bad, your 6-hour print wanders into the garbage can after having completed 5 hours flawlessly.
With a sensor setup, the print could have been paused, you could have found out what was wrong, fix it and maybe save the print.

mjh11

jon-bondy, 2015-03-03:
I have to say that I agree with mjh.  Feedback systems on our printers 
(including position sensors for the extruder) would allow faster speeds 
and more reliable printing.  From time to time, one of my hobbed gears 
loads up with plastic.  It would be great if the software said "time to 
clean the gear" rather than my discovering this because prints are 
failing.  Yes, I can inspect this once a week, but I do not see this 
feature request as being unreasonable.

Jon

kyrreaa, 2015-03-03:
This is what I have been working on for a bit. (And what I have been
talking about on IRC.)

The hobbed bolts are good over a certain range but will deviate. I have
done quite a lot of testing on different hobbed bolts and other
feeders/extruders and also looked into the filament diameter as source to
some discrepancys I found in my prints. In the end it turns out that 1.75mm
filament atleast (which is all I ever use nowadays) are very good in
dimensions (atleast from my suppliers) and there was very little room for
improvement here. The feed however varied quite a lot, a factor 20 vs the
filament diameter when it comes to total volume influence.

I've been testing multiple optical sensors and writing interface code to
use a few of them. The plan is to add support in my branch of Repetier fw
to test it on a live printer.
I am having some issues with the communication though due to the way the
entire software is structured and timed. It does not lend itself easily to
such quick mods where you need to add extra communication on a new bus.

Anyway, in case others want to play I can share some fun info.
First off, development the last years have made many of the optical sensors
into a single chip solution. They don't just integrate the optical sensing
elements and the correlator anymore.  Some even have USB or wireless more
or less integrated. The cheap chips we can easily get seem to end up as
USB. This ofcource is a problem for our poor controllers but it turns out
they still support PS/2 which is much easier and run over the D+/D- signals
of the USB plug. I've been adding PS/2 code to allow these to be read.
There are still chips out there that are SPI and I2C too, so we do have
options.

SPI would ofcource be preferred due to speed and better noise immunity, but
in Atmel chips this is still a poosly supported peripheral and will take
time away from other tasks.

As for the actual sensors, yes they are mostly automatic and they do
integrate motion all by themselves. The integrals reset every time you read
them and there are overflow-flags to signal you if you are reading them too
infrequently.
A very cool feature of some of the sensors is ofcource that we can even
read out the image array.  This could theoretically be used to estimate
filament width using some edge estimation techniques and taking advantage
of the pixel depth available in the sensor to gain sub-pixel accuracy. This
technique works great for LIDAR systems so should be ok for filament
estimation too.

For me, this is just another thing I fiddle with in a sea of stuff I need
to get done so I am ofcource pressed for time. So much cool stuff to do, so
little time...
I wouldn't expect my stuff to be ready anytime soon, but I do have around
20 more optical chips on the way for more experiments. I plan to make a
sensorboard that will integrate easily and add to the bowden extruders in a
easy way.