repetier/Repetier-Host #331: Feature request:Pause/Continue command from printer/serial

luc-github, 2015-03-05:
Today, there are several commands from printer to host : start, wait, ok, temperatures, files list, etc...
to stop a host print from printer it just need to send "start" command on serial.because if host receive a "start"it reset/stop current Job.

It would be really nice if there is a "pause" command, the printer send "pause" to host by serial then host react like there is a @pause in GCODE and pause the current job

this is useful when host and printer are not in same room and need to pause to change filament or fix issue and ou are in front of printer, no need to go back to host then back to printer. just press button which send "pause" to host. 
This also would allow to ask a pause to host if printer detect no more filament or a Jam 

In same spirit would be great if have also a command "continue" when host is in pause, printer send "continue" in serial then host resume from pause, like if user press resume on host.

repetier, 2015-03-05:
That is already implemented. If firmware sends

RequestPause:

host will start a pause like if you have hit pause button.

Newest firmware ha salso a filament change routine which stops host printing (by not accepting new commands) until change is finished. This even works with any host software and with sd printing.

Jam detection will work the same once it is working as supposed.

closed 2015-03-05 by repetier

luc-github, 2015-03-05:
Thanks a lot - that is great - may I know where this "RequestPause:" is documented or refered ? - So I can check in the future, because I was looking around before but did not found - that is why I raised this  feature request.

Thanks again

repetier, 2015-03-05:
I guess it is not documented:-) 

It's in ui.cpp used for pausing the host:

        case UI_ACTION_PAUSE:
            Com::printFLN(PSTR("RequestPause:"));
            break;

luc-github, 2015-03-05:
Arf I missed it in FW....
Thanks again for your time