repetier/Repetier-Host #335: Cannot move objects using the mouse

MachineKing, 2015-07-04:
Cannot move objects using the mouse.
I have always been able to move objects by holding right mouse button. Now I can't.
Nothing changed, I tried to move an object after completing a print and it doesn't work.

repetier, 2015-07-04:
You can only move in the object placement tab! If you have more then one object selected click right next to an object to move all objects.

Darker, 2017-09-10:
Well it doesn't work very reliable on the placement tab. I just googled this issue because I can't move those objects.

repetier, 2017-09-10:
Just mark them so a box around them appears. And in rotation mode right click and drag in 3d window.

Darker, 2017-09-10:
I know how to do it. I found a model that, when added to the board, blocks moving of all objects. I'll start a new issue and will upload the model.

Darker, 2017-09-10:
Seems that the issue is caused by lag that occurs before every drag attempt. If the lag takes long enough, user gives up and assumes that the model cannot be moved.

I just tried to reproduce issue I had and suddenly I could move the model.

repetier, 2017-09-10:
So your model is so big that painting takes more the 0.05 seconds? Should normally not happen with nowadays computers using VBOs in graphic card for acceleration. Do you have enabled VBOs for drawing or do you need one of the slow methods?