repetier/Repetier-Host #337: start up error

adasown, 2015-07-28:
what‘s wrong with this issue
CLR20r3
01:	repetierhost.exe
02:	1.0.0.0
03:	55b091c5
04:	OpenTK
05:	1.1.804.3121
06:	4f61c964
07:	570
08:	8e
09:	System.AccessViolationException
OS :	6.1.7601.2.1.0.768.2
ID:	2052
1:	2e99
2:	2e9966990555f9ffbdbec265e5203498
3:	3583
4:	358338b9acf01c7f7b27ec94ba8c7963

MarceloTetraBoard, 2016-01-27:
Hello, My name is Marcelo. I install Repetier Host and this message appears at the end of the installation.

C:\Program Files\Repetier-Host

Descrição:
  Stopped working

Assinatura do problema:
  Nome do Evento de Problema:	CLR20r3
  Assinatura do Problema 01:	repetierhost.exe
  Assinatura do Problema 02:	1.0.0.0
  Assinatura do Problema 03:	511694c8
  Assinatura do Problema 04:	RepetierHost
  Assinatura do Problema 05:	1.0.0.0
  Assinatura do Problema 06:	511694c8
  Assinatura do Problema 07:	f0
  Assinatura do Problema 08:	363
  Assinatura do Problema 09:	System.InvalidCastException
  Versão do sistema operacional:	6.1.7601.2.1.0.256.1
  Identificação da Localidade:	1046

My i7 machine - 32Ram - Win7

repetier, 2016-01-27:
Is latest .NET installed and did you check with .net repair tool if there is something wrong? This is somewhere in the used .NET runtime libraries and normally does not happen.