repetier/Repetier-Host #339: EEPROM and M501 don't report the home offsets

NotDrexin, 2015-09-16:
As stated in the title whenever i check EEPROM or M501, M212, or home offsets that I have set in Marlin, do not appear. Everyting else in M501 and EEPROM appear except that. EEPROM will save the offset until I reconnect and disconnect too. My printer is a Migbot, and I have repetier version 1.5.5 and my configuration.f is as follows

```gcode
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD 33
 //   #define MOTHERBOARD BOARD_ULTIMAKER
#endif

// Define this to set a custom name for your generic Mendel,
// #define CUSTOM_MENDEL_NAME "This Mendel"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
// #define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

// This defines the number of extruders
#define EXTRUDERS 1

//// The following define selects which power supply you have. Please choose the one that matches your setup
// 1 = ATX
// 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)

#define POWER_SUPPLY 1

// Define this to have the electronics keep the power supply off on startup. If you don't know what this is leave it.
// #define PS_DEFAULT_OFF

//===========================================================================
[... 686 lines cut ...]
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H
```

repetier, 2015-09-16:
Hi,

as I understand you you can set home offsets in eeprom settings in host but
they do not get displayed,right? In this case please provide a log about
the data Marlin (which version btw?) provides. If marlin has changed format
sinc eimplementation this might happen.

Yours
Roland Littwin

-- 
Mit freundlichen Grüßen

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in
Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

NotDrexin, 2015-09-17:
Hey. thanks for getting back to me. I can change settings in the eeprom, but they do not display and are not carried over to the next time my printer reconnects to host. I'm using V1.5.5

```log
16:38:44.760 : echo:Hardcoded Default Settings Loaded
16:38:44.760 : echo:Steps per unit:
16:38:44.761 : echo:  M92 X80.00 Y80.00 Z398.27 E94.50
16:38:44.761 : echo:Maximum feedrates (mm/s):
16:38:44.761 : echo:  M203 X500.00 Y500.00 Z5.00 E25.00
16:38:44.761 : echo:Maximum Acceleration (mm/s2):
16:38:44.761 : echo:  M201 X8000 Y8000 Z100 E10000
16:38:44.761 : echo:Acceleration: S=acceleration, T=retract acceleration
16:38:44.761 : echo:  M204 S3000.00 T3000.00
16:38:44.761 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
16:38:44.761 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
16:38:44.761 : echo:Home offset (mm):
16:38:44.761 : echo:  M206 X0.00 Y0.00 Z0.00
16:38:44.761 : echo:PID settings:
16:38:44.761 : echo:   M301 P22.20 I1.08 D114.00
```

intomotion, 2015-09-17:
Vanaf 19 september t/m 28 september ben ik op vakantie. In deze periode ben ik beperkt bereikbaar.
Als u een storing wilt melden stuur dan een SMS bericht naar 0626781755.

Met vriendelijke groet,
Johan van Leur
Into-Motion.

repetier, 2015-09-18:
I see firmware sending

 echo:Home offset (mm):
16:38:44.761 : echo:  M206 X0.00 Y0.00 Z0.00

so it is correct to show 0 in host. It will show what firmware says. So if Marlin does not show it it is a firmware error I guess, except you had defined it to 0 here, but that would be a bad example showing it.

closed 2016-02-07 by NotDrexin