repetier/Repetier-Host #343: No start signal detected - forcing start

Blastin06, 2016-03-08:
New office computer, installed serial_install (same one I used on my old computer that worked) Installed Repetier. Computer connects in Repetier, shows green connected button, but displays the No start signal detected - forcing start dialog.  Device manager shows that it is on Com 4 and displays it as a USB Serial Port.   Co-worker shares the 3D printer with me and everything shows the same (device manager, port settings, printer settings) except for it works on his computer.

What am I missing? Why can I no longer use my printer? Why am I getting the No start signal?  Any direction is appreciated!

repetier, 2016-03-08:
Typically it is wrong baud rate or protocol in printer settings. If you can not connect at all it might be the server you instaled and configured which autoconnects then blocking the port.

Blastin06, 2016-03-08:
On my Co-workers computer it works on 115200 and autodetect. The same settings do not work on my computer.  I'm not sure what you mean by the server. I have tried V1.6.1 and V1.5.6  If it helps, Im using windows 7 64bit.

DLTAssets, 2016-06-10:
Did you find a solution for this?  I'm experiencing the same thing today.  The printer worked well for past few months and now it doesn't.  

I use a RepRap Melzi ATMEGA1284P (This looks like the model http://www.ebay.co.uk/itm/3D-Printer-Circuit-Board-Control-Panel-For-RepRap-Melzi-ATMEGA1284P-MK1-MK2a-MK3-/291409716218?hash=item43d95f63fa:g:4n0AAOSwPhdVCG8b)

I do not have the server on and when I try connect with Repetier I see these warnings/errors:-

23:38:04.369 : No start signal detected - forcing start
23:38:14.200 : No start signal detected - forcing start
23:38:37.243 : No start signal detected - forcing start
23:38:50.343 : No start signal detected - forcing start
23:39:17.646 : No start signal detected - forcing start
23:40:00.782 : Communication timeout - reset send buffer block

Thanks!

repetier, 2016-06-11:
One possible reason might be FTDI fake chips. FTDI does not like them and das different drivers making them unusable or changing data by adding errors if it detects a fake chip. So if your driver updated lately that might be the problem. Google "ftdi fake" for more informations about driver versions doing this.

DLTAssets, 2016-06-11:
Wow!  Nasty people!  Using malware to punish victims of fraud, changes not for the perpetrators the reward!