repetier/Repetier-Host #344: Time prediction way of

frank16, 2016-05-10:
My carthesian printer has a fairly heavy table. To get nice edges I only have a printing acceleration of 150mm/s^2 and travel acceleration of 300mm^2.
I use Cura engine most. Unfortunately it is internally configured at 3000mm/s^. I think this is why my predicted print time is way of, by a factor of 4 or so (seems to be a complaint of many people).
Would it be possible to bring this value out to the cura engine configuration in repetier host?
I had a look at it myself, but I can't even find the interface script for cura engine :-(
I could change cura engine for my printer and re-make, but that is such an ugly solution...

repetier, 2016-05-11:
Don't look at curaengine computed time. In host printer settings you can set a factor. If you use the repetier-server as printing daemon it will even take correct accelerations into account giving even more accurate estimates.

frank16, 2016-05-18:
O.k, so there are ways around it. But can't it be fixed properly? I'm willing to put in some time and try coding it myself, but at least would you point me to the repetier-host to cura-engine interface? There is too much stuff and no documentation...

repetier, 2016-05-18:
Repetier-Host is closed source. This version here is antique. The versions we deliver are far ahead of this. So nothing you can do.

frank16, 2016-05-19:
Snif :-(

closed 2016-05-19 by frank16