repetier/Repetier-Host #345: printed object ~120 degrees rotated versus what repetier host shows

tiagom62, 2016-05-14:
I am using Repetier-Host V1.6.1 and Marlin 1.1.0-RC6.

I have my tower set exactly how is shows here:

http://forums.reprap.org/addon.php?178,module=embed_images,url=https%3A%2F%2Fc2.staticflickr.com%2F4%2F3857%2F15264990135_a0fc011cfc_b.jpg

X is the left tower Y is the right tower and Z is the back tower.

When i import an object, slice it with cura and then print it, what repetier shows versus what is actually printed is about 120 degrees off.

When i use the manual controls, left/right moves the hotend left/right, up/down moves the hotend front/back. 

Any hints on how i can resolve this? thanks

repetier, 2016-05-14:
Have you oriented the view in host the same using the coordinate axis? Default view is rotated.

Otherwise it makes no sense if moves are executed correctly. Especially 120° are hard to get - normally would be 90°/270°/180° if axis get flipped. Host just send what you see in preview and coordinates are interpreted by firmware and you said it interprets correctly so it must also print correctly.

tiagom62, 2016-05-20:
looks like it was user error, thanks for the prompt reply!

closed 2016-05-20 by tiagom62