repetier/Repetier-Host #348: Err exception : InvalidCastException exception

daspapiya, 2016-07-05:
when i am building the software it is showing me this exception :- 

An unhandled exception of type 'System.InvalidCastException' occurred in RepetierHost.exe

Additional information: Unable to cast object of type 'System.Int32' to type 'System.String'.

Please figure it out whats the matter ..

repetier, 2016-07-05:
Please download the current version at www.repetier.com, this version is not maintained for years and expired.

closed 2016-07-05 by repetier

daspapiya, 2016-07-05:
can i get the source code of the current version
its urgent

repetier, 2016-07-05:
It's not open source, so not really:-(

We offer custom versions if you need changes and also allow plugins in current versions if you want to modify something yourself like adding wizards etc. You then still need custom version to create a package for delivery, but personal use is free of course.

daspapiya, 2016-07-05:
ok thanks

daspapiya, 2016-07-05:
i was just trying up adding up some functionality, needed  the source for repetier , if you can provide, even a earlier version
@repetier