repetier/Repetier-Host #349: Filament jam detection commands

karimcitoh, 2016-07-11:
I have a question about how to interact with the filament jam detection situation.

When the filament detector enters a paused status to replace the filament, the only way I have to resume the print, is by physically using the standard display.

In the serial communication logs I find this messages from the printer:

```
Received from printer: "important:Extruder jam detected"
Received from printer: "Info:Marked all extruders as unjammed."
Received from printer: "Jam control disabled:1"
Received from printer: "// action:pause"
```

Any idea where I can find the commands to set the temperature and resume the print when this mode is entered. 

Printer does not seem to react against any GCODE sent via USB. Only interaction via the standard LCD with wheel.

Thanks in advance

repetier, 2016-07-12:
Problem is to decide who should take care of filament change. You selected LCD as option, so it stops and blocks communication or commands would get lost and then it makes no sense. So yes, all you can do is reheat and extruder by turning wheel.

Alternatively you can set to request pause from host. Works with Repetier-Host and Server. If you use a host not understanding that request it will simply continue to print. Drawback is that it takes a bit longer to pause, but then you can use manual controls to do more things.

karimcitoh, 2016-07-12:
Thank you very much for your answer!

So the only thing I need is to not select LCD as filament change option, in order to have serial communications available when this happens. I guess I can do this when building the firmware.

And then, what commands should I use (send to the printer) to deal with this situation? Is the interaction flow somewhere documented?

Again, thanks for your help.

repetier, 2016-07-12:
>
> Thank you very much for your answer!
>
> So the only thing I need is to not select LCD as filament change 
> option, in order to have serial communications available when this 
> happens. I guess I can do this when building the firmware.
Yes, it is one of the options what to do on jam detection.
>
> And then, what commands should I use (send to the printer) to deal 
> with this situation? Is the interaction flow somewhere documented?
It just should cause a pause from host side then. So you can do whatever 
needed to fix and then unpause.

karimcitoh, 2016-07-12:
Again, sorry for opening this issue here. I think the right place was Repetier-firmware, and not host.

I am building software to interact with my 3D printer, so it's something that interacts directly with the firmware, and not with host.
My question is then: what does the printer send over the serial communication when there is extruder jamming. And what are the commands to pause/unpause the 3D printer when this happens. Since I need to build functionality to interact with this situation.

Thanks for your patience.

repetier, 2016-07-12:
From Printer.cpp jam event

```
    case PRINTER_INTERRUPT_EVENT_JAM_DETECTED:
        EVENT_JAM_DETECTED;
        Com::printFLN(PSTR("important:Extruder jam detected"));
        UI_ERROR_P(Com::translatedF(UI_TEXT_EXTRUDER_JAM_ID));
#if JAM_ACTION == 1 // start dialog
        Printer::setUIErrorMessage(false);
#if UI_DISPLAY_TYPE != NO_DISPLAY
        uid.executeAction(UI_ACTION_WIZARD_JAM_EOF, true);
#endif
#elif JAM_ACTION == 2 // pause host/print
#if SDSUPPORT
        if(sd.sdmode == 2) {
            sd.pausePrint(true);
            break;
        }
#endif // SDSUPPORT
        Com::printFLN(PSTR("RequestPause:Extruder Jam Detected!"));
#endif // JAM_ACTION
        break;

```

So with JAM_ACTION 2 it just sends
RequestPause:Extruder Jam Detected!

Everything else has to be done by host. On host you just hit continue to continue printing after you changed filament(fixed jam.