repetier/Repetier-Host #350: Question about UI

luc-github, 2016-07-24:
Hi,
I have a project using ESP8266 as wifi bridge with printer (https://github.com/luc-github/ESP3D)
the device allow to monitor and control the printer using web interface.
I did a repetier style UI, still ongoing  
[image]

Is that ok for you ? I do not any see license issue as no code from you (all from me and subject to improvement ^_^) but I prefer to ask and  you may be interested for repetier server.   

Here a the current test page before integration you can play with overlay, sliders and buttons:
https://github.com/luc-github/ESP3D/blob/master/UI/testui.htm
I tested it on Chrome/Opera/Internet Explorer/Microsoft Edge/Firefox/Safari 5 (windows) / Safari 9 (OSX)
Here the thread of test/progress.
https://github.com/luc-github/ESP3D/issues/50

I just wanted to be sure there is no issue from your side before continuing.
Please let me know and feel free to comment.

Thanks

repetier, 2016-07-25:
Also it is not very creative to copy existing solutions 1:1 it is ok, as you are not using our images directly (I see you have recreated them looking just very similar) and code. So go on:-)

closed 2016-07-25 by repetier

luc-github, 2016-07-25:
Thanks - actually it will be one UI among others to show how UI can be customized, not necessary the main one.