repetier/Repetier-Host #354: Pressing Axes Movement Increments in Print Panel displays twice the Value (Mac / RH 1.00)

mriveralee, 2016-08-04:
Steps to repro:
1) Connect printer
2) Move increment "1" on y-axis 
3) Witness the display shows "2.00"
4) Move increment "100" on x-axis
5) Witness display shows "200"

--
Thanks for checking all these bugs out :)

repetier, 2016-08-05:
Sorry, but I can not reproduce this problem. Moving code has not changed for years and still does work as expected, at least for me. Please check your log what gets send. Could it be that you double clicked the move buttons?

mriveralee, 2016-08-10:
The output logs show that the proper commands are being sent. So it seems like a UI bug.
Here's a video of the repro: https://dl.dropboxusercontent.com/u/21057133/steps_multiple_of_2_bug.mov

Below are the corresponding logs for the video:
```
> 1:31:28 PM: N104 G28 X0.00 *80
< 1:31:29 PM: X:0.00 Y:0.00 Z:0.000 E:0.0000
> 1:31:30 PM: N107 G91 *55
> 1:31:30 PM: N108 G1 X1.00 F4800.000 *18
> 1:31:30 PM: N109 G90 *56
> 1:31:34 PM: N114 G91 *53
> 1:31:34 PM: N115 G1 X10.00 F4800.000 *46
> 1:31:34 PM: N116 G90 *54
> 1:31:38 PM: N121 G91 *51
> 1:31:38 PM: N122 G1 X100.00 F4800.000 *26
> 1:31:38 PM: N123 G90 *48
> 1:31:41 PM: N127 G28 X0.00 *81
< 1:31:45 PM: X:0.00 Y:0.00 Z:0.000 E:0.0000
> 1:31:47 PM: N132 G91 *49
> 1:31:47 PM: N133 G1 Y0.10 F4800.000 *27
> 1:31:47 PM: N134 G90 *54
> 1:31:50 PM: N138 G91 *59
> 1:31:50 PM: N139 G1 Y1.00 F4800.000 *17
> 1:31:50 PM: N140 G90 *53
> 1:31:52 PM: N143 G91 *55
> 1:31:52 PM: N144 G1 Y10.00 F4800.000 *43
> 1:31:52 PM: N145 G90 *48
> 1:31:54 PM: N148 G91 *60
> 1:31:54 PM: N149 G1 Y100.00 F4800.000 *22
> 1:31:54 PM: N150 G90 *52
> 1:31:57 PM: N154 G91 *49
> 1:31:57 PM: N155 G1 Y-100.00 F4800.000 *54
> 1:31:57 PM: N156 G90 *50
> 1:31:59 PM: N158 G91 *61
> 1:31:59 PM: N159 G1 Y-100.00 F4800.000 *58
> 1:31:59 PM: N160 G90 *55
> 1:32:00 PM: N162 G28 Y0.00 *81
< 1:32:01 PM: X:0.00 Y:0.00 Z:0.000 E:0.0000
> 1:32:02 PM: N166 G91 *48
> 1:32:03 PM: N167 G1 Z1.000 F100.000 *20
> 1:32:03 PM: N168 G90 *63
> 1:32:05 PM: N171 G28 Z0.000 *96
< 1:32:12 PM: X:0.00 Y:0.00 Z:0.000 E:0.0000
```

repetier, 2016-08-16:
Found reason and fixed it for 1.0.2 coming today.

closed 2016-08-16 by repetier