repetier/Repetier-Host #355: Linux: Startup crash

pwnage-pineapple, 2016-09-18:
Here is the log written to stdout when I ran repetierHost from the command-line:

`(process:13814): Gtk-WARNING **: Locale not supported by C library.
	Using the fallback 'C' locale.

(mono:13814): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(mono:13814): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
libpng warning: iCCP: known incorrect sRGB profile
[... 82 lines cut ...]
  at System.Windows.Forms.ScrollableControl.OnLayout (System.Windows.Forms.LayoutEventArgs levent) <0x40db3c10 + 0x00057> in <filename unknown>:0 
  at System.Windows.Forms.ContainerControl.OnLayout (System.Windows.Forms.LayoutEventArgs e) <0x40db3be0 + 0x00013> in <filename unknown>:0 
  at System.Windows.Forms.Control.PerformLayout (System.Windows.Forms.Control affectedControl, System.String affectedProperty) <0x40d9b710 + 0x00288> in <filename unknown>:0 
  at System.Windows.Forms.Control.OnResizeInternal (System.EventArgs e) <0x40d9b660 + 0x0002b> in <filename unknown>:0 
  at System.Windows.Forms.Control.OnResize (System.EventArgs e) <0x40d9b630 + 0x00018> in <filename unknown>:0 
  at System.Windows.Forms.UserControl.OnResize (System.EventArgs e) <0x40dfad60 + 0x00013> in <filename unknown>:0 
  at System.Windows.Forms.Control.OnSizeChanged (System.EventArgs e) <0x40d9b520 + 0x00032> in <filename unknown>:0 
  at System.Windows.Forms.Control.UpdateBounds (Int32 x, Int32 y, Int32 width, Int32 height, Int32 clientWidth, Int32 clientHeight) <0x40d9b030 + 0x001f3> in <filename unknown>:0 
  at System.Windows.Forms.Control.UpdateBounds (Int32 x, Int32 y, Int32 width, Int32 height) <0x40da8e60 + 0x000bf> in <filename unknown>:0 
  at System.Windows.Forms.Control.SetBoundsCoreInternal (Int32 x, Int32 y, Int32 width, Int32 height, BoundsSpecified specified) <0x40da8950 + 0x0037f> in <filename unknown>:0 
  at System.Windows.Forms.Control.SetBoundsCore (Int32 x, Int32 y, Int32 width, Int32 height, BoundsSpecified specified) <0x40da8910 + 0x0002d> in <filename unknown>:0 
  at System.Windows.Forms.Control.SetBoundsInternal (Int32 x, Int32 y, Int32 width, Int32 height, BoundsSpecified specified) <0x40da1ba0 + 0x00143> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control:SetBoundsInternal (int,int,int,int,System.Windows.Forms.BoundsSpecified)
  at System.Windows.Forms.Layout.DefaultLayout.LayoutDockedChildren (System.Windows.Forms.Control parent, System.Windows.Forms.Control[] controls) <0x40d9c1e0 + 0x00557> in <filename unknown>:0 
  at System.Windows.Forms.Layout.DefaultLayout.Layout (System.Object container, System.Windows.Forms.LayoutEventArgs args) <0x40d9bef0 + 0x000ef> in <filename unknown>:0 
  at System.Windows.Forms.Control.OnLayout (System.Windows.Forms.LayoutEventArgs levent) <0x40d9bc80 + 0x0011a> in <filename unknown>:0 
  at System.Windows.Forms.ScrollableControl.OnLayout (System.Windows.Forms.LayoutEventArgs levent) <0x40db3c10 + 0x00057> in <filename unknown>:0 
  at System.Windows.Forms.Control.PerformLayout (System.Windows.Forms.Control affectedControl, System.String affectedProperty) <0x40d9b710 + 0x00288> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control:PerformLayout (System.Windows.Forms.Control,string)
  at System.Windows.Forms.Control+ControlCollection.Add (System.Windows.Forms.Control value) <0x40df51a0 + 0x00aef> in <filename unknown>:0 
  at System.Windows.Forms.TabPage+TabPageControlCollection.Add (System.Windows.Forms.Control value) <0x40ea0150 + 0x00013> in <filename unknown>:0 
  at RepetierHost.Main.InsertComponents () <0x4100cbf0 + 0x001ea> in <filename unknown>:0 
  at RepetierHost.Main..ctor () <0x40d82ba0 + 0x03c8f> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.Main:.ctor ()
  at RepetierHost.Program.Main (System.String[] args) <0x40d3a1e0 + 0x00037> in <filename unknown>:0 `

repetier, 2016-09-19:
Looks like a problem in mono forms library while computing scrollbar position. In which Linux/Mono combination does this happen?

pwnage-pineapple, 2016-09-19:
Arch Linux with kernel 4.7.4 and mono 4.4.1.0

bethebeer, 2016-12-22:
I am having a similar issue using ubuntu 16.04 and mono 4.2.1
mono -V outputs this: 
`Mono JIT compiler version 4.2.1 (Debian 4.2.1.102+dfsg2-7ubuntu4)
`
and trying to open repetierHost from a command line causes an outline of the startup window to appear, then dissapear, and then this error:

```
System.InvalidOperationException: Cannot call Invoke or BeginInvoke on a control until the window handle is created
  at System.Windows.Forms.Control.FindControlToInvokeOn () <0x41a03ba0 + 0x000df> in <filename unknown>:0 
  at System.Windows.Forms.Control.Invoke (System.Delegate method, System.Object[] args) <0x41a03ae0 + 0x0001f> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control:Invoke (System.Delegate,object[])
  at System.Windows.Forms.Control.Invoke (System.Delegate method) <0x41a03950 + 0x000bb> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control:Invoke (System.Delegate)
  at RepetierHost.view.utils.WebInfoPanel.ReadFromDisk () <0x41991530 + 0x0089f> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) <0x7f3dde4b1010 + 0x00099> in <filename unknown>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x7f3dde4af850 + 0x0016e> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x7f3dde4af820 + 0x00020> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) <0x7f3dde4af770 + 0x00059> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart () <0x7f3dde4b11d0 + 0x0002e> in <filename unknown>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.InvalidOperationException: Cannot call Invoke or BeginInvoke on a control until the window handle is created
  at System.Windows.Forms.Control.FindControlToInvokeOn () <0x41a03ba0 + 0x000df> in <filename unknown>:0 
  at System.Windows.Forms.Control.Invoke (System.Delegate method, System.Object[] args) <0x41a03ae0 + 0x0001f> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control:Invoke (System.Delegate,object[])
  at System.Windows.Forms.Control.Invoke (System.Delegate method) <0x41a03950 + 0x000bb> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.Windows.Forms.Control:Invoke (System.Delegate)
  at RepetierHost.view.utils.WebInfoPanel.ReadFromDisk () <0x41991530 + 0x0089f> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) <0x7f3dde4b1010 + 0x00099> in <filename unknown>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x7f3dde4af850 + 0x0016e> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x7f3dde4af820 + 0x00020> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) <0x7f3dde4af770 + 0x00059> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart () <0x7f3dde4b11d0 + 0x0002e> in <filename unknown>:0 
```

Whytehorse, 2017-01-22:
Same here, crash on startup. My error is a little different:
`Stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) OpenTK.Platform.Egl.Egl.GetCurrentContext () <0xffffffff>
  at OpenTK.Platform.Egl.Egl.get_IsSupported () <0x00013>
  at OpenTK.Platform.Factory..cctor () <0x00167>
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) <0xffffffff>
  at <unknown> <0xffffffff>
  at OpenTK.Graphics.GraphicsMode..cctor () <0x0004f>
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) <0xffffffff>
  at <unknown> <0xffffffff>
  at RepetierHostExtender.utils.RHOpenGL..ctor () <0x0002f>
  at (wrapper remoting-invoke-with-check) RepetierHostExtender.utils.RHOpenGL..ctor () <0xffffffff>
  at RepetierHost.view.ThreeDControl.InitializeComponent () <0x006d3>
  at RepetierHost.view.ThreeDControl..ctor () <0x0059b>
  at (wrapper remoting-invoke-with-check) RepetierHost.view.ThreeDControl..ctor () <0xffffffff>
  at RepetierHost.Main..ctor () <0x01d0f>
  at (wrapper remoting-invoke-with-check) RepetierHost.Main..ctor () <0xffffffff>
  at RepetierHost.Program.Main (string[]) <0x00037>
  at (wrapper runtime-invoke) <Module>.runtime_invoke_void_object (object,intptr,intptr,intptr) <0xffffffff>

Native stacktrace:

	mono() [0x49ff2f]
	mono() [0x4f390e]
	mono() [0x426c19]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fec2c604390]

Debug info from gdb:

[New LWP 28032]
[New LWP 28033]
[New LWP 28034]
[New LWP 28035]
[New LWP 28036]
[New LWP 28039]
[New LWP 28041]
[New LWP 28042]
[New LWP 28043]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007fec2c603f7b in __waitpid (pid=28045, stat_loc=0x7fec2d14910c, options=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
29	../sysdeps/unix/sysv/linux/waitpid.c: No such file or directory.
  Id   Target Id         Frame 
* 1    Thread 0x7fec2d121780 (LWP 28031) "mono" 0x00007fec2c603f7b in __waitpid (pid=28045, stat_loc=0x7fec2d14910c, options=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
  2    Thread 0x7fec2b7ff700 (LWP 28032) "mono" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  3    Thread 0x7fec2bf4a700 (LWP 28033) "Finalizer" 0x00007fec2c602827 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0, futex_word=0x956820) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  4    Thread 0x7fec1a732700 (LWP 28034) "dconf worker" 0x00007fec2c31db5d in poll () at ../sysdeps/unix/syscall-template.S:84
  5    Thread 0x7fec19f31700 (LWP 28035) "gmain" 0x00007fec2c31db5d in poll () at ../sysdeps/unix/syscall-template.S:84
  6    Thread 0x7fec19730700 (LWP 28036) "gdbus" 0x00007fec2c31db5d in poll () at ../sysdeps/unix/syscall-template.S:84
  7    Thread 0x7fec18a72700 (LWP 28039) "Timer-Scheduler" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  8    Thread 0x7fec18377700 (LWP 28041) "Timer-Scheduler" __clock_nanosleep (clock_id=<optimized out>, flags=1, req=0x7fec18376db0, rem=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:48
  9    Thread 0x7fec18336700 (LWP 28042) "Threadpool work" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  10   Thread 0x7fec03fff700 (LWP 28043) "Threadpool work" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185

Thread 10 (Thread 0x7fec03fff700 (LWP 28043)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x000000000058a192 in ?? ()
#2  0x0000000000587b2b in ?? ()
#3  0x0000000000629a8c in ?? ()
#4  0x00007fec2c5fa6ba in start_thread (arg=0x7fec03fff700) at pthread_create.c:333
#5  0x00007fec2c32982d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 9 (Thread 0x7fec18336700 (LWP 28042)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x000000000058a192 in ?? ()
#2  0x0000000000587b2b in ?? ()
#3  0x0000000000629a8c in ?? ()
#4  0x00007fec2c5fa6ba in start_thread (arg=0x7fec18336700) at pthread_create.c:333
[... 21 lines cut ...]
#9  0x00007fec2baa84f8 in ?? ()
#10 0x0000000000000051 in ?? ()
#11 0x00007fec04000e00 in ?? ()
#12 0x0000000000000051 in ?? ()
#13 0x00007fec18a718d0 in ?? ()
#14 0x00007fec18a71840 in ?? ()
/build/gdb-cXfXJ3/gdb-7.11.1/gdb/dwarf2-frame.c:690: internal-error: Unknown CFI encountered.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/build/gdb-cXfXJ3/gdb-7.11.1/gdb/dwarf2-frame.c:690: internal-error: Unknown CFI encountered.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================
`

martin31821, 2017-03-10:
Same problem here:
* Arch
* Linux 4.10.1
* Mono 4.8.0 (e4a3cf3)

No further output beyond the ones posted here.

Iambecomeroot, 2017-09-06:
Not sure if it's the same issue but I'm getting [this stacktrace](https://pastebin.com/XfSYbTdC).

* Linux Zeus 4.12.8-2-ARCH #1 SMP PREEMPT Fri Aug 18 14:08:02 UTC 2017 x86_64 GNU/Linux
* mono 5.0.0.100-2

repetier, 2017-09-06:
@Iambecomeroot This is an error during open gl window initialization inside the OpenTK library. Since that is not my code it is hard to say what is going wrong here. I guess the library is not getting one of the expected responses from OpenGL and then does something wrong causing the crash. Unfortunately nothing I can improve unless there comes an update of the library that handles it.

If you have no OpenGL with true color configured for linux that would also explain the problem

Iambecomeroot, 2017-09-06:
> If you have no OpenGL with true color configured for linux that would also explain the problem

I am using the nvidia driver if that changes anything. I don't know what you mean by configuring opengl.

repetier, 2017-09-07:
I meant changing color to true color but that should be default by all current computers. Not sure what else could make a difference here. Not really a linux desktop user, more command line and on my desktop with ati card it still works. So I can not really say why OpenTK and your OpenGL implementation do not seem to play well.

Iambecomeroot, 2017-09-07:
I didn't change anything but when I turned on my computer this morning, it worked. Weird considering I've been having this problem for a while and through multiple reboots. Anyway, thanks for trying to help.

Whytehorse, 2017-09-07:
It's working now for me too. It must have been a bug in one of the libraries used and it was fixed upstream

repetier, 2017-09-08:
Guess so. Some automatic update having fixed it. Closing issue.

closed 2017-09-08 by repetier