repetier/Repetier-Host #358: Ability to recognize mk4duo as fw.

MagoKimbra, 2017-02-10:
I apologize for this request, but it may be possible that repetier host to recognize mk4duo as fw to enable read and write EEPROM functions.
In addition the possibility to enable the printer 3D map function.
Basically if I change in command M115 the written FIRMWARE_NAME: MK4duo with FIRMWARE_NAME: repetier it all works, I just have to adjust a few things, but most work.
Thank you so much and sorry again.

repetier, 2017-02-10:
I don't have this on my printers, so I have to ask back. So far I was assuming it is a marlin clone with improvements. If I read
https://github.com/MagoKimbra/MK4duo/blob/master/MK4duo/src/eeprom/eeprom.cpp
that shows so many commands that modify eeprom and do not use the repetier system where one command handles all eeprom values. So I wonder how that is working if you set repetier in firmware string.  That makes host think it is repetier and uses that command.

Also 3d map function, did you test it with "repetier" trick? This greatly depends on output format like repetier-firmware uses it. So if you have tested it I could also aktivate it for that firmware.

MagoKimbra, 2017-02-10:
I tried it and it works, for the EEPROM if you tell me that's one thing that you can do I try to bring the most similar to that of writing can reptier firmware.
The only differences are that I have the steps, the feedrate, acceleration and jerk divided by extruder, but that part could also remain excluded. It would be nice to have things in common.

repetier, 2017-02-10:
Not sure where you split the 2 questions.
Just read your G30 and it has no handling of P1,P2,P0 parameter which is used in repetier to activate/deactivate or keep state. Of course ignoring these is ok if it is activated before.
So if you mean yes, output is recogniced correctly showing height map, that is great. Then I can add it to whitelist for that feature. If you tell me I need to send special commands to activate/deactivate z probe I can also add these as special case.

Regarding eeprom, I have to say the marlin way is a pain for host developers. We have a minimal eeprom support for marlin. Main problem I have with this is that they change output and there is no way I know to detect which values which version of firmware supports.  So as soon as I ask or modify it to include a new eeprom value others scream that it does not work.

Repetier solution is so simple and flexible that I do not understand why they do not use it. We only need 2 commands and they never change for any update. They automatically support all added features as M205 simple returns currently set values with descriptions and M206 allows changing every byte in eeprom.

- M205 - Output EEPROM settings
- M206 - Set EEPROM value

So I think that part is not working with your firmware?

Since you are more the expert for marlin, do you know a documentation that describes how to see which eeprom commands are supported and if the response has changed over time? Maybe there is a simple test to just send a bunch of commands to query the parameter and if we get a known answer we add them. Similar to what we do for repetier as a editable table. That way we could make it work somehow for marlin/kimbra.

BTW: Is there a difference in commands/responses between MK4due and the marlin kimbra? In no, is it to be expected or do features get added always to both.

MagoKimbra, 2017-02-10:
[image]
This I did with Mk4duo you see is fine..

For the EEPROM it is not very clear as you use it in your fw. If you allow me to do so I can see how you use you ...
Marlin or Mk4duo it works so M503 returns the current values:

```
20:19:23.135 : Config:Steps per unit:
20:19:23.155 : Config:  M92 X160.000 Y160.000 Z160.000 E796.000
20:19:23.155 : Config:Maximum feedrates (mm/s):
20:19:23.155 : Config:  M203 X200.000 Y200.000 Z200.000 E45.000
20:19:23.155 : Config:Maximum Acceleration (mm/s2):
20:19:23.155 : Config:  M201 X3000 Y3000 Z3000 E1500
20:19:23.155 : Config:Accelerations: P=printing, V=travel and T* R=retract
20:19:23.155 : Config:  M204 P3000.000 V3000.000 R10000.000
20:19:23.156 : Config:Advanced variables: S=Min feedrate (mm/s), V=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum X jerk (mm/s), Y=maximum Y jerk (mm/s), Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
20:19:23.156 : Config:  M205 S0.000 V0.000 B20000 X10.000 Y10.000 Z10.000 E10.000
20:19:23.156 : Config:Home offset (mm):
20:19:23.156 : Config:  M206 X0.000 Y0.000 Z0.000
20:19:23.156 : Config:Endstop adjustment (mm):
20:19:23.156 : Config:  M666 X-0.303250 Y0.000000 Z-0.162256
20:19:23.187 : Config:Geometry adjustment: ABC=TOWER_RADIUS_ADJ, IJK=TOWER_POSITION_ADJ, UVW=TOWER_DIAGROD_ADJ, R=Delta Radius, D=Diagonal Rod, S=Segments per second, H=Z Height
20:19:23.187 : Config:  M666 A-0.438 B0.581 C0.000 I0.000 J0.000 K0.000 U0.000 V0.000 W0.000 R109.359680 D213.476531 S400.000 H216.460
20:19:23.187 : Config:Z Probe offset (mm):
20:19:23.187 : Config:  M666 P-2.468618
20:19:23.187 : Config:PID settings:
20:19:23.187 : Config:  M301 H0 P19.891265 I0.931776 D106.158172
20:19:23.187 : Config:Filament settings: Disabled
20:19:23.187 : Config:  M200 D1.750
20:19:23.187 : Config:Motor current:
20:19:23.188 : Config:  M906 X1.00 Y1.00 Z1.00 E1.30
20:19:23.188 : Stats: Total: 532, Finished: 489, Failed: 43
20:19:23.188 : Stats: Total print time: 19d 23h 4m 43s, Power on time: 13d 21h 12m 30s
20:19:23.188 : Stats: Filament used: 2Km 907m 45cm 9mm
```

M500 write all values in EEPROM.

MagoKimbra, 2017-02-10:
Ok I understand ... In Mk Marlin or the EEPROM is written sequentially and depending on how the firmware is set, then in the same eeprom position there is never the same as between a firmware and the other ...
While in repetier it was previously held the positions of each single figure with the definitions of EPR_.
In this way with the M206 command you can go to type directly into the EEPROM of position because it is safe to hold that data.
This could be a right way I could go.
However, this means that to be totally compatible with repetier I must necessarily use the same repetier positions. As the baud rate does not record it in EEPROM, however, I have to consider cmq that cell for baud rate ..
Or I can give you a table but is always that you will use to manage the eeprom MK4duo?

repetier, 2017-02-11:
The position in eeprom is completely up to you the repetier way. M205 returns something like this excerpt:

8:24:44.072: EPR:0 1028 0 Language
8:24:44.077: EPR:2 75 250000 Baudrate
8:24:44.077: EPR:3 129 0.000 Filament printed [m]
8:24:44.077: EPR:2 125 0 Printer active [s]
8:24:44.081: EPR:2 79 1200000 Max. inactive time [ms,0=off]
8:24:44.081: EPR:2 83 360000 Stop stepper after inactivity [ms,0=off]
8:24:44.085: EPR:3 3 610.0000 X-axis steps per mm
8:24:44.085: EPR:3 7 610.0000 Y-axis steps per mm
8:24:44.087: EPR:3 11 6400.0000 Z-axis steps per mm
8:24:44.088: EPR:3 15 200.000 X-axis max. feedrate [mm/s]
8:24:44.088: EPR:3 19 200.000 Y-axis max. feedrate [mm/s]

First value is data type, second is eeprom position and 3rd is value. The rest is description to show users in editor with units in [] so we can put them behind value in editor.

I think I will try to fix marlin eeprom using the M503. Whatever it shows and is recogniced will be put into a table together with a command to set the value. So I can use most of repetier method but convert it from marlin. As you see M205 is already used for marlin so using it double is a bit difficult and after all it is marlin and saying marlin with repetier eeprom. So we keep it compatible, also more work for me. Will update my avr marlin printer to latest version to see if it works.

MagoKimbra, 2017-02-11:
So if I do I implement the command 205 and 206, but making my own table, you can use it safely in host?

repetier, 2017-02-11:
Yes. that work if I handle the exception that marlin kimbra uses repetier eeprom handling. But that would make all older versions not working. Which is why I said I'd better try convertig it host side like you have it now. So using M503 values to convert them. That way old kimbra and marlin will benefit from it.

repetier, 2017-02-14:
One quick question. I have changed firmware detector to detect mk4due as marlin derivative so I could test eeprom editor available already. It did not detect anything since your eeprom response is
config:
and in marlin this used to be
echo:

Is that something you changed to be more incompatible or did the marlin guys introduce the change? In theory this is a good idea to distinguish it from the many echo responses, on the other hand it slows down hosts as they need to compare now 2 possible solutions.

MagoKimbra, 2017-02-14:
No this is my choice to make it clear to the user that the data configuration.

repetier, 2017-02-15:
Ok, i have modified eeprom editor to now map commands to a table, so we can simply show eeprom variables we detect. Mapping is done with a xml file that can easily be extended if new eeprom parameter appear. Also your firmware now maps to marlin firmware and enables bed height map.

From the writing of the xml file I learned that you changed M301 from Ex to Hx. While such changes might seem nice when chars have a easier meaning also E for extruder of H for heater makes no big difference to me, this makes your firmware incompatible and forces all existing hosts to write new exceptions for your firmware. If that doe snot happen your users will suffer the consequences and even if it happens, they will have to wait until the hosts get modified. So I think it is better to stick to existing standards and only add new stuff that is not present so it is not possible otherwise. Just what I think, but it is your firmware:-)

closed 2017-02-15 by repetier

MagoKimbra, 2017-02-16:
I made some changes because I think the right, for example in the case of a multicolored hotend as the diamond, extruders are 3, but the hotend is 1. I think it is wrong to make M301 E0 because the user could even think about writing M301 E1 since it has three extruders. At the beginning each extruder corresponded a hotend today it is no longer so, I think it's time to diversify extruders from hotend.
However for me the priority now is to be compatible as possible with the EEPROM reading repetier, and if I have to change certain commands and functions do it.
Perhaps with the M205 command that tells the values repetier list I can just as you indicated, but when the user write command M503 i write M301 H.
They are available to any of your choice ...

repetier, 2017-02-16:
Just to be clear. I have it now implemented so please do not change again:-) I use M503 as I do for stock marlin and it works.

MagoKimbra, 2017-02-16:
Ok thank's!!!

MagoKimbra, 2017-04-22:
Hi thank's i donwload new repetier host.. Very nice...
I do not want to break again, if you tell me or have a table of how to recognize the data I fit ...
With M92 Xxxx Yyyy Zzzz Eeee I see that it reads only the steps for mm of the three axes, but not of E. The same thing for feedrates, accelerations etc etc.
If you tell me how the fields must be compiled I modify the firmware for the new version ...
Thank you and sorry again ...

repetier, 2017-04-23:
Ok, that is easy as I made a config file for new entries, so if someone misses a entry he can add it and send me the new missing entries for next update. Check in
C:\Program Files\Repetier-Host\plugins\RRFirmwares\marlineeprom.xml
and see if it contains all cases you have. Tried my best but seems I overlooked some. Just mail me the additions I missed and I add them on next update that comes soon I think (when slic3r 1.3.0 final is out).

MagoKimbra, 2017-04-23:
Thank's thank's and thank's....
I've already changed the file and I'm ok then I'm sending you the file.
Only problem with negative numbers, I do not know very well XML so I do not know how to solve ...

```
<!-- Z Probe offset -->
  <entry type="float">
    <descripton unit="[unit]">Z Probe offset</descripton>
    <set>M666 P@</set>
    <detect>M666.*P(\d*\.?\d*)</detect>
  </entry>
```
But i read the cell in red..
M666 P-2.3 this is my  number...

Can you help me??

repetier, 2017-04-23:
\d is digit 0-9 so you need -? before it if it can be negative

```
  <entry type="float">
    <descripton unit="[unit]">Z Probe offset</descripton>
    <set>M666 P@</set>
    <detect>M666.*P(-?\d*\.?\d*)</detect>
  </entry>
```

These are simple regualr expressions and you can test them for example here:
https://regex101.com/

MagoKimbra, 2017-04-23:
Perfect thank you very much!
Now it works perfectly ...

MagoKimbra, 2017-09-24:
Sorry if I re-open, but why does not the new version recognize the response to the M114 command?
I checked with your firmware, but I do not think there are any differences.
I respond to M114 as follows:
X: 0.00 Y: 0.00 Z: 200.150 E: 0.0000
But nothing, does not put the values in the coordinates ... I'm wrong in the protocol?
Thank's

repetier, 2017-09-24:
Host is not interpreting any M114 command, that is not marlin kimbra specific. Problem is that newer  Marlin versions started handling inclusion of G92 offsets differently, so interpretation might be wrong. Also it is dangerous to update coordinates from M114 since that might already be expired by following move commands.

MagoKimbra, 2017-09-24:
You are right, but after making the home on the delta I move down to a safe zone, then after the home release the new coordinates as a response. But on a host I keep seeing the maximum height and not the actual one. With the old repetier, the coordinate values were no longer updated. To resynchronize the two things, you have to give a G1 Zzzz command so that both on repetier and the printer I see the same height ...

repetier, 2017-09-24:
Changing Z heights are in deed a small problem here. You can set them in host as homing position and then you have the right coordinates. But after a calibration this changes and needs to be readjusted.

For normal prints this does not really make a difference as there are global coordinates used and from there on they are in sync.

MagoKimbra, 2017-09-24:
You are right about this, but it seemed helpful for the repetier host to update the coordinates when it received a response to m114 as it does for temperatures.
Cmq ok, I thought it was a problem of my firmware ...
thanks again for your availability

repetier, 2017-09-24:
Directly after homing is the only time where it makes sense. Problem is to decide if any move commands have be send since then. Maybe I find a way to improve this as it makes also problems for us for getting questions like this. On the other side, we got also complaints that M114 confused the coordinates so this is a delicate problem.