repetier/Repetier-Host #361: Slicer configuration button: Unknown option: gui-mode

glaudiston, 2017-04-09:
I'm using a GNU/Linux-x86_64

My Repetier-Host version is V1.6.2

I  load the slicer gui with "--gui" there is no "--gui-mode" in Slic3r.

All seems to work as expected except I can't open slic3r configuration from slicer tab:

```
18:29:50.835 : <Slic3r> Unknown option: gui-mode
18:29:50.838 : <Slic3r> Slic3r 1.3.0-dev is a STL-to-GCODE translator for RepRap 3D printers
18:29:50.838 : <Slic3r> written by Alessandro Ranellucci <aar@cpan.org> - http://slic3r.org/
18:29:50.838 : <Slic3r> Usage: slic3r.pl [ OPTIONS ] [ file.stl ] [ file2.stl ] ...
18:29:50.838 : <Slic3r>     --help              Output this usage screen and exit
18:29:50.838 : <Slic3r>     --version           Output the version of Slic3r and exit
18:29:50.838 : <Slic3r>     --save <file>       Save configuration to the specified file
18:29:50.838 : <Slic3r>     --load <file>       Load configuration from the specified file. It can be used 
18:29:50.838 : <Slic3r>                         more than once to load options from multiple files.
18:29:50.838 : <Slic3r>     -o, --output <file> File to output gcode to (by default, the file will be saved
18:29:50.838 : <Slic3r>                         into the same directory as the input file using the
18:29:50.838 : <Slic3r>                         --output-filename-format to generate the filename.) If a
18:29:50.838 : <Slic3r>                         directory is specified for this option, the output will
18:29:50.838 : <Slic3r>                         be saved under that directory, and the filename will be
18:29:50.838 : <Slic3r>                         generated by --output-filename-format.
18:29:50.838 : <Slic3r>   
18:29:50.838 : <Slic3r>   Non-slicing actions (no G-code will be generated):
18:29:50.838 : <Slic3r>     --repair            Repair given STL files and save them as <name>_fixed.obj
18:29:50.838 : <Slic3r>     --cut <z>           Cut given input files at given Z (relative) and export
18:29:50.838 : <Slic3r>                         them as <name>_upper.stl and <name>_lower.stl
18:29:50.838 : <Slic3r>     --split             Split the shells contained in given STL file into several STL files
18:29:50.838 : <Slic3r>     --info              Output information about the supplied file(s) and exit
18:29:50.838 : <Slic3r>     
18:29:50.838 : <Slic3r>     -j, --threads <num> Number of threads to use
18:29:50.838 : <Slic3r>   GUI options:
18:29:50.838 : <Slic3r>     --gui               Forces the GUI launch instead of command line slicing (if you
18:29:50.838 : <Slic3r>                         supply a model file, it will be loaded into the plater)
18:29:50.838 : <Slic3r>     --autosave <file>   Automatically export current configuration to the specified file
18:29:50.838 : <Slic3r>   Output options:
18:29:50.838 : <Slic3r>     --output-filename-format
18:29:50.838 : <Slic3r>                         Output file name format; all config options enclosed in brackets
18:29:50.838 : <Slic3r>                         will be replaced by their values, as well as [input_filename_base]
18:29:50.838 : <Slic3r>                         and [input_filename] (default: [input_filename_base].gcode)
18:29:50.838 : <Slic3r>     --post-process      Generated G-code will be processed with the supplied script;
18:29:50.838 : <Slic3r>                         call this more than once to process through multiple scripts.
18:29:50.838 : <Slic3r>     --export-svg        Export a SVG file containing slices instead of G-code.
18:29:50.838 : <Slic3r>     -m, --merge         If multiple files are supplied, they will be composed into a single 
18:29:50.838 : <Slic3r>                         print rather than processed individually.
18:29:50.838 : <Slic3r>   
18:29:50.838 : <Slic3r>   Printer options:
18:29:50.839 : <Slic3r>     --nozzle-diameter   Diameter of nozzle in mm (default: 0.5)
18:29:50.839 : <Slic3r>     --print-center      Coordinates in mm of the point to center the print around
18:29:50.839 : <Slic3r>                         (default: 100,100)
18:29:50.839 : <Slic3r>     --z-offset          Additional height in mm to add to vertical coordinates
18:29:50.839 : <Slic3r>                         (+/-, default: 0)
18:29:50.839 : <Slic3r>     --z-steps-per-mm    Number of full steps per mm of the Z axis. Experimental feature for
18:29:50.839 : <Slic3r>                         preventing rounding issues.
18:29:50.839 : <Slic3r>     --gcode-flavor      The type of G-code to generate (reprap/teacup/repetier/makerware/sailfish/mach3/machinekit/smoothie/no-extrusion,
18:29:50.839 : <Slic3r>                         default: reprap)
18:29:50.839 : <Slic3r>     --use-relative-e-distances Enable this to get relative E values (default: no)
18:29:50.839 : <Slic3r>     --use-firmware-retraction  Enable firmware-controlled retraction using G10/G11 (default: no)
18:29:50.839 : <Slic3r>     --use-volumetric-e  Express E in cubic millimeters and prepend M200 (default: no)
18:29:50.839 : <Slic3r>     --gcode-arcs        Use G2/G3 commands for native arcs (experimental, not supported
18:29:50.839 : <Slic3r>                         by all firmwares)
18:29:50.839 : <Slic3r>     --gcode-comments    Make G-code verbose by adding comments (default: no)
18:29:50.839 : <Slic3r>     --vibration-limit   Limit the frequency of moves on X and Y axes (Hz, set zero to disable;
18:29:50.839 : <Slic3r>                         default: 0)
18:29:50.839 : <Slic3r>     --pressure-advance  Adjust pressure using the experimental advance algorithm (K constant,
18:29:50.839 : <Slic3r>                         set zero to disable; default: 0)
18:29:50.839 : <Slic3r>     
18:29:50.839 : <Slic3r>   Filament options:
[... 215 lines cut ...]
18:29:50.843 : <Slic3r>                         Set a different extrusion width for top infill
18:29:50.843 : <Slic3r>     --support-material-extrusion-width
18:29:50.844 : <Slic3r>                         Set a different extrusion width for support material
18:29:50.844 : <Slic3r>     --infill-overlap    Overlap between infill and perimeters (default: 55%)
18:29:50.844 : <Slic3r>     --bridge-flow-ratio Multiplier for extrusion when bridging (> 0, default: 1)
18:29:50.844 : <Slic3r>   
18:29:50.844 : <Slic3r>    Multiple extruder options:
18:29:50.844 : <Slic3r>     --extruder-offset   Offset of each extruder, if firmware doesn't handle the displacement
18:29:50.844 : <Slic3r>                         (can be specified multiple times, default: 0x0)
18:29:50.844 : <Slic3r>     --perimeter-extruder
18:29:50.844 : <Slic3r>                         Extruder to use for perimeters and brim (1+, default: 1)
18:29:50.844 : <Slic3r>     --infill-extruder   Extruder to use for infill (1+, default: 1)
18:29:50.844 : <Slic3r>     --solid-infill-extruder   Extruder to use for solid infill (1+, default: 1)
18:29:50.844 : <Slic3r>     --support-material-extruder
18:29:50.844 : <Slic3r>                         Extruder to use for support material, raft and skirt (1+, default: 1)
18:29:50.844 : <Slic3r>     --support-material-interface-extruder
18:29:50.844 : <Slic3r>                         Extruder to use for support material interface (1+, default: 1)
18:29:50.844 : <Slic3r>     --ooze-prevention   Drop temperature and park extruders outside a full skirt for automatic wiping
18:29:50.844 : <Slic3r>                         (default: no)
18:29:50.844 : <Slic3r>     --standby-temperature-delta
18:29:50.844 : <Slic3r>                         Temperature difference to be applied when an extruder is not active and
18:29:50.844 : <Slic3r>                         --ooze-prevention is enabled (default: -5)
18:29:50.844 : <Slic3r>     

```

repetier, 2017-04-10:
When we wrote 1.6.2 slic3r 1.2.9 was the newest version and there the option was --gui-mode. Looks like this was now changed making it incompatible with old calls. So we will add a new version in manager for next release to select --gui for next release.

Thanks for the hint.

glaudiston, 2017-04-10:
Glad to know will be fixed.
Until then, I did create a patch to slic3r.pl:
```
30c30
<         'gui'                   => \$opt{gui},
---
>         'gui|gui-mode=s'          => \$opt{gui},
```
With it i'm able to open slic3r via configuration button, but i need to open the file manually from `~/.local/share/RepetierHost/composition.stl`

repetier, 2017-04-10:
It is actually composition.amf that gets created when you try to slice with slic3r. stl is used for CuraEngine.