repetier/Repetier-Host #362: SD Upload wouldn't complete

yannick9906, 2017-04-15:
I couldn't find a similiar issue anywhere, so here it is:

Everytime I try to upload gcode via the host it uploads fine, but stops after about 300kB of Data. No Error is displayed. The actual gcode would be 8,560,640 bytes big, but only 369242 bytes were uploaded.
Everytime I retry uploading the same file another 100kB of additional data is uploaded.
Are there some limitiations on how big a uploaded file can be?

repetier, 2017-04-16:
No, there is no known limitation. You should try reformatting sd card with sd formater. Might be a problem there that stops or something with the communication is not stable enough. Can you use firmware normally after it stops uploading? Any hints in log?

Also I would normally copy such long files directly on the oc, that is so much faster that it is even faster when you need to remove sd card 2 times.

yannick9906, 2017-04-16:
I formated the 4gb SD Card using SD Formatter and tried again.
Here's the log: (It stopped after about 8% -> ~400kB uploaded)
```
*** File Data ***
13:49:14.964 : Uploading time: 17674429h:49m:14s
13:49:14.981 : error writing to file
13:49:14.981 : N23436 M29*24
13:49:15.006 : N23437 M105*38
13:49:15.113 : Done saving file.
13:49:15.758 : N23438 M20*31
13:49:15.760 : Begin file list
13:49:15.770 : TOTEM 395877
13:49:15.772 : End file list
```
I can still use the firmware normally.

> Also I would normally copy such long files directly on the oc, that is so much faster that it is even faster when you need to remove sd card 2 times.

That's what I usually do, but I wanted to give it a try...

iMogal, 2017-04-16:
If this is a 'cheap' SD card, that maybe the problem. Have you tried another known to be good card?

repetier, 2017-04-16:
"Done saving file" comes from firmware when it received a M29

This might also be the result of a communication error.
Can you enable echo and log the upload. Then you should see around the upload stop what host send and what firmware thinks it received.

yannick9906, 2017-04-17:
Here the log with all Commands and Echos:
```
17:27:29.096 : N210363 G1 X101.157 Y107.196 E354.4724*107
17:27:29.096 : N210364 G1 X100.901 Y107.208 E354.4767*101
17:27:29.096 : N210365 G1 X100.665 Y107.216 E354.4806*110
17:27:29.702 : Uploading time: 17674457h:27m:29s
17:27:29.719 : error writing to file
17:27:29.720 : ok 13752
17:27:29.720 : N210366 M29*40
17:27:29.720 : N210367 M105*22
17:27:29.768 : ok 13753
17:27:29.768 : ok 13754
17:27:29.768 : ok 13755
17:27:29.772 : ok 13756
17:27:29.772 : ok 13757
17:27:29.772 : ok 13758
17:27:29.780 : Done saving file.
17:27:29.785 : ok 13759
17:27:29.787 : T:31.97 /0 B:54.98 /55 B@:163 @:0
17:27:29.935 : N210368 M20*47
17:27:29.940 : ok 13760
17:27:29.940 : Begin file list
17:27:29.945 : EAGLE 245349
17:27:29.945 : SOLDIER.GCO 1294336
17:27:29.947 : End file list
17:27:30.939 : wait
17:27:31.938 : wait
17:27:32.780 : N210369 M105*24
17:27:32.783 : ok 13761
17:27:32.789 : T:31.97 /0 B:55.17 /55 B@:0 @:0
17:27:33.782 : wait
17:27:34.785 : wait
17:27:35.784 : wait
17:27:35.842 : N210370 M105*16
17:27:35.846 : ok 13762
17:27:35.849 : T:31.62 /0 B:55.17 /55 B@:0 @:0
17:27:36.846 : wait
17:27:37.849 : wait
17:27:38.849 : wait
17:27:38.902 : N210371 M105*17
17:27:38.906 : ok 13763
17:27:38.909 : T:31.28 /0 B:55.17 /55 B@:0 @:0
17:27:39.905 : wait
17:27:40.905 : wait
17:27:41.908 : wait
17:27:41.969 : N210372 M105*18
17:27:41.974 : ok 13764
17:27:41.977 : T:31.11 /0 B:55.05 /55 B@:0 @:0
17:27:42.973 : wait
17:27:43.972 : wait
17:27:44.976 : wait
17:27:45.026 : N210373 M105*19
17:27:45.029 : ok 13765
17:27:45.036 : T:30.94 /0 B:55.05 /55 B@:153 @:0
17:27:46.028 : wait
17:27:47.032 : wait
17:27:48.031 : wait
17:27:48.092 : N210374 M105*20
17:27:48.097 : ok 13766
17:27:48.099 : T:30.60 /0 B:55.05 /55 B@:233 @:0
17:27:49.096 : wait
17:27:50.095 : wait
17:27:51.099 : wait
17:27:51.152 : N210375 M105*21
17:27:51.157 : ok 13767
17:27:51.159 : T:30.34 /0 B:55.17 /55 B@:52 @:0
17:27:52.156 : wait
17:27:53.155 : wait
17:27:54.158 : wait
17:27:54.216 : N210376 M105*22
17:27:54.220 : ok 13768
17:27:54.221 : T:30.26 /0 B:55.11 /55 B@:0 @:0
17:27:55.220 : wait
17:27:56.219 : wait
17:27:57.223 : wait
17:27:57.279 : N210377 M105*23
17:27:57.284 : ok 13769
17:27:57.286 : T:30.00 /0 B:55.23 /55 B@:42 @:0
17:27:58.283 : wait
17:27:59.282 : wait
17:28:00.286 : wait
17:28:00.340 : N210378 M105*24
17:28:00.344 : ok 13770
17:28:00.346 : T:29.88 /0 B:55.23 /55 B@:0 @:0
17:28:01.343 : wait
17:28:02.342 : wait
17:28:03.346 : wait
17:28:03.404 : N210379 M105*25
17:28:03.407 : ok 13771
17:28:03.409 : T:29.51 /0 B:55.17 /55 B@:52 @:0
17:28:04.407 : wait
17:28:05.410 : wait
17:28:06.409 : wait
17:28:06.468 : N210380 M105*31
17:28:06.471 : ok 13772
17:28:06.477 : T:29.39 /0 B:55.05 /55 B@:233 @:0
17:28:07.470 : wait
17:28:08.473 : wait
17:28:09.481 : wait
17:28:09.533 : N210381 M105*30
17:28:09.535 : ok 13773
17:28:09.541 : T:29.02 /0 B:55.23 /55 B@:42 @:0
17:28:10.534 : wait
17:28:11.537 : wait
17:28:12.536 : wait
17:28:12.593 : N210382 M105*29
17:28:12.599 : ok 13774
17:28:12.601 : T:28.90 /0 B:55.23 /55 B@:42 @:0
17:28:13.597 : wait
17:28:14.597 : wait
17:28:15.601 : wait
17:28:15.653 : N210383 M105*28
17:28:15.658 : ok 13775
17:28:15.660 : T:28.66 /0 B:54.98 /55 B@:163 @:0
```

@iMogal My SD Card is a Kingston Class 4 SDHC 4GB Card... Should be a good card, but maybe some blocks are broken... Edit: No Damaged Blocks were found.
I tried with another card(Panasonic Class 2 SDHC 4GB) and the upload worked w/o any problems.
It's currently printing this file from the SD Card, but it should finish normally.
I'll close this issue as soon as the print finishes.

repetier, 2017-04-17:
As "error writing to file" indicates the fat library returned an error while writing. Since I do not understand how that works it is just a fact that something bad happened and firmware stopped upload for that reason.

SD Cards have spi mode and a parallel mode that normally gets used. Printers use SPI mode as it requires less pins and is still fast enough. Quality of that might vary or it is just something other on the card. I always use cheap cards with no problems.

yannick9906, 2017-04-18:
Okay, the print finished as it should. So it's maybe just a broken SD Card.

closed 2017-04-18 by yannick9906