repetier/Repetier-Host #363: Repetier-Host 2.0.1 Fails to print

greenaar, 2017-05-16:
Windows 10 (multiple computers tried).
Repetier host 2.0.0 and 2.0.1 tried.
Connection to printer directly over usb. 
Connection to repetier server .80.3

Error: 13:28:33.265 : Error:expected line 111 got 256
13:28:33.276 : Resend:111
13:28:33.283 : Error:expected line 111 got 256
13:28:33.284 : Resend:111
13:28:36.291 : Warning: Seems like we missed a ok, got a wait - continue sending.

(Note, the line number varies).

This occurs at both 115200 baud and 250000 baud.

The only common point I can see here is both printers I've tried are running Repetier Firmware 1.0.0dev. 

Please note, this whole configuration works perfectly on 1.6.2, and fails 100% of the time on 2.x.

repetier, 2017-05-17:
You mention repetier server - so are you printing ove rlocal server connected to usb or do you ignore server and connect directly with host?

Can you post a except of real log before error starts to see where it comes from. A difference in line numbers of 145 is a bit big so I wonder where the difference comes from and why it was not detected before knowing that it worked for my installations.

greenaar, 2017-05-22:
Sorry for the delay getting back to you.

I've tried both printing to repetier server over the network (which is connected via a Pi, and works properly on 1.62 but fails on 2.0.1) and direct via usb using repetier host with no repetier server installed (which also fails on 2.x).  Note the log section above is from a direct connection, and occurs immediately after the heating stage (those were the first messages once the hotend reached temperature).

There is no movement prior to the errors listed.