repetier/Repetier-Host #364: Does EEPROM setting for filament diameter overrider slicer's gcode?

Ianmcmill, 2017-06-03:
Hello,

I just switched from Repetier Host 1.6 to 2.0.1 and started printing. I realized that there was a big over-extrusion. I wanted to calibrate the extruder but the measured data are odd. I issued to extrude 50mm but measured that around 88mm were extruded.
I opened the EEPROM settings in Repetier and realized, that there is a "new" setting called 
`Filament diameter extr. 0 = 3.00mm` 
but I am using 1.75mm and a modified multiplier in Slic3r.
Two questions:
1. Is this a Marlin EEPROM setting? 
I have never seen this before in repetier 1.6.x and never used it.
Now I am confronted with it and I don't know how to deal with it. Set with M200 D1.75? But is it then cubic milimeters or linear mm units?
2. How do I disable this?
M200 D0 is supposed to "go back to linear mm units" according to the wiki. But when issuing this and saving EEPROM with M500, the EEPROM still shows the previously set 1.75mm respectively 3.00mm from the default settings.

I was not sure where to post this issue, marlin repository or repetier's but as this setting just came up with Repetier 2.0.1 (#better Marlin EEPROM handling in changelog) I decided to post this here.

repetier, 2017-06-03:
These are parameter we get from marlin firmware when we query for eeprom settings. So it is not directly host related, except that we can now display it as eeprom setting.

I never use volumetric extrusion, but I think these values are just some defaults in case you would enable volumetric extrusion. Also host would not change/send command to marlin if you do not change the value. Only modified values get send when hitting "ok" button.

Please also have a look at the log. there you might see more informations as we only extract parts we know and omit comments. So it might show if volumetric extrusion is enabled.

Ianmcmill, 2017-06-03:
Ok thank you.
I think this might have something to do with `M502 `I used before. This would restore firmware defaults and maybe somehow enabled the "Filament diameter" setting. 
Not sure though but as of now this is the only logic explanation I can find.