repetier/Repetier-Host #365: disable G104 command

mkeyno, 2017-06-07:
hi , I used repetier to send the g-code to my 3D printer with Marline firmware , however I've tried to used specific version of Marlin for CNC engraving laser, however the repetier host repeatedly send G104 command to get the temperature whereas there is no temp sensor in the  CNC engraving laser and hence the communication timeout or reset , the question is how to prevent repetier  to send G104 command?
BR

repetier, 2017-06-07:
We do not send G104. We send M105 for temperature polling and that can be disabled in Printer Settings menu. Just disable "Check Bed and Extruder Temperature".

However if your engraver software adds commands these may come from there.

mkeyno, 2017-06-07:
thanks @repetier  I found  it , however don't you have any plan to made official branch of repetier host for cnc laser target ?

repetier, 2017-06-08:
No, no plans for this. Will see when we have finished the more urgent features in our products to add optimized views for this.

mkeyno, 2017-06-08:
thanks anyway

closed 2017-06-08 by mkeyno