repetier/Repetier-Host #367: Crash when moving cursor over empty grid

zaventh, 2017-07-28:
Affects at least Repetier 2.0.1 and 2.0.2. When the application is first loaded and the grid is empty, moving the cursor around in the empty grid for more than 1-2 seconds causes this crash. If I skirt around the grid and just hit the menus and go to manual control, I can avoid the crash.

Details below. Full stacktrace attached.

```
$ mono RepetierHost.exe
System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at RepetierHostExtender.utils.MoveUpDown.MoveUpDown_MouseMove (System.Object sender, System.Windows.Forms.MouseEventArgs e) [0x00105] in <62f229d6f5264485b42f8edbdffa4398>:0 
  at System.Windows.Forms.Control.OnMouseMove (System.Windows.Forms.MouseEventArgs e) [0x00019] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.Control.WmMouseMove (System.Windows.Forms.Message& m) [0x000a5] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.Control.WndProc (System.Windows.Forms.Message& m) [0x0020c] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ScrollableControl.WndProc (System.Windows.Forms.Message& m) [0x00000] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.Control+ControlWindowTarget.OnMessage (System.Windows.Forms.Message& m) [0x00000] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.Control+ControlNativeWindow.WndProc (System.Windows.Forms.Message& m) [0x0000b] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.NativeWindow.WndProc (System.IntPtr hWnd, System.Windows.Forms.Msg msg, System.IntPtr wParam, System.IntPtr lParam) [0x00085] in <c86f0be4f9a847d3b8d63306221e6732>:0 
```

```
$ mono --version
Mono JIT compiler version 5.0.1.1 (2017-02/5077205 Thu May 25 09:19:18 UTC 2017)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	LLVM:          supported, not enabled.
	GC:            sgen (concurrent by default)
```

```
$ lsb_release -a && uname -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:printing-9.20160110ubuntu0.2-amd64:printing-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	LinuxMint
Description:	Linux Mint 18.2 Sonya
Release:	18.2
Codename:	sonya
Linux sol 4.10.17-041017-generic #201705201051 SMP Sat May 20 14:53:33 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
[rep_host_crash.txt](https://github.com/repetier/Repetier-Host/files/1182088/rep_host_crash.txt)

repetier, 2017-07-28:
When you say grid do you mean move controls? From the error it is a problem in the moveupdown control but I would associate grid more with the bed.

zaventh, 2017-07-28:
I mean the "3D view" pane with the default isometric view of the bed grid. I can't reproduce it reliably but typically I run the app, click Connect, then if my cursor moves across the "3D View" pane on my way to the manual control tab, many times it crashes right there. It never crashes if I avoid this 3D view area.

If I load an STL or a GCODE file and am in Object Placement or Print Preview, I don't think it has ever happened FWIW. It seems only when it's empty. It does sometimes happen though after loading a GCODE, printing it, then if my mouse intersects the Print Preview 3D View pane it will crash with the same error. So the emptiness of the 3D view pane may be a red herring.

repetier, 2017-07-28:
Very strange. Since it crashes in a move event in up/down control I would not have guessed it getting any data when not over that control. Will analyse the control and add some constrains if I see it would be possible to get out of range.

repetier, 2017-07-28:
Ok, I could not reproduce it but found the position where the error came front and found at least a theoretical possibility to get that error under some circumstances. Have fixed it, so please download current 2.0.2 again and let me know if the fix worked.

zaventh, 2017-07-31:
This issue appears to be fixed in at least v2.0.5. Thanks!

closed 2017-07-31 by zaventh