repetier/Repetier-Host #368: Windows, print preview broken in v2.0.2

4tmorris, 2017-07-30:
Windows 7, since v2.0.2
Parts of the g code are not showing in the print preview. 
[image]

repetier, 2017-07-30:
If you used Slic3r Prusa Engine they are simply not there. We reverted back to 1.36.0 which does slice it correctly. SO please update to 2.0.4

4tmorris, 2017-08-01:
That is what I am seeing with v2.0.4. with gcode generated by slic3r PE 1.36.0. 
[image]

Here is the same gcode displayed by v2.0.1 
[image]

repetier, 2017-08-01:
We found a visualization bug. It is already fixed in 2.0.5 and now we also include Slic3r PE 1.36.2. There we could not reproduce it any more, so hopefully completely fixed.

closed 2017-08-01 by repetier