repetier/Repetier-Host #370: Cannot open application in Fedora 26

loi219, 2017-08-22:
I've installed the application via dnf and when I'm trying to open it, I have this error: ``` WARNING: The runtime version supported by this application is unavailable.
Using default runtime: v4.0.30319 ```

```
Unhandled Exception:
System.InvalidCastException: Specified cast is not valid.
  at RepetierHost.connector.SerialConnector.LoadFromRegistry () [0x000d6] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at RepetierHost.view.FormPrinterSettings.comboConnector_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x0007f] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at System.Windows.Forms.ComboBox.OnSelectedIndexChanged (System.EventArgs e) [0x00023] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.SetSelectedIndex (System.Int32 value, System.Boolean supressAutoScroll) [0x000ab] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.set_SelectedIndex (System.Int32 value) [0x00000] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at RepetierHost.view.FormPrinterSettings.load (System.String printername) [0x00097] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at RepetierHost.view.FormPrinterSettings.comboPrinter_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x0001e] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at System.Windows.Forms.ComboBox.OnSelectedIndexChanged (System.EventArgs e) [0x00023] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.SetSelectedIndex (System.Int32 value, System.Boolean supressAutoScroll) [0x000ab] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.set_SelectedIndex (System.Int32 value) [0x00000] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.set_Text (System.String value) [0x00081] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at RepetierHost.view.FormPrinterSettings.load (System.String printername) [0x0000c] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at RepetierHost.view.FormPrinterSettings..ctor () [0x000db] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.view.FormPrinterSettings:.ctor ()
  at RepetierHost.Main..ctor () [0x0019c] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.Main:.ctor ()
  at RepetierHost.Program.Main (System.String[] args) [0x00010] in <cb8865b345a24ae6b571400be0f920a2>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.InvalidCastException: Specified cast is not valid.
  at RepetierHost.connector.SerialConnector.LoadFromRegistry () [0x000d6] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at RepetierHost.view.FormPrinterSettings.comboConnector_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x0007f] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at System.Windows.Forms.ComboBox.OnSelectedIndexChanged (System.EventArgs e) [0x00023] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.SetSelectedIndex (System.Int32 value, System.Boolean supressAutoScroll) [0x000ab] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.set_SelectedIndex (System.Int32 value) [0x00000] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at RepetierHost.view.FormPrinterSettings.load (System.String printername) [0x00097] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at RepetierHost.view.FormPrinterSettings.comboPrinter_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x0001e] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at System.Windows.Forms.ComboBox.OnSelectedIndexChanged (System.EventArgs e) [0x00023] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.SetSelectedIndex (System.Int32 value, System.Boolean supressAutoScroll) [0x000ab] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.set_SelectedIndex (System.Int32 value) [0x00000] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at System.Windows.Forms.ComboBox.set_Text (System.String value) [0x00081] in <2f8b04a5fbc8420bbb3895e91a62c74f>:0 
  at RepetierHost.view.FormPrinterSettings.load (System.String printername) [0x0000c] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at RepetierHost.view.FormPrinterSettings..ctor () [0x000db] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.view.FormPrinterSettings:.ctor ()
  at RepetierHost.Main..ctor () [0x0019c] in <cb8865b345a24ae6b571400be0f920a2>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.Main:.ctor ()
  at RepetierHost.Program.Main (System.String[] args) [0x00010] in <cb8865b345a24ae6b571400be0f920a2>:0
```

repetier, 2017-08-23:
Did you use our AppImage as adviced? This includes all libraries required and worked for me also under fedora.

loi219, 2017-08-27:
I can open it but it's impossible to connect with the printer. With cura the printer is connected