repetier/Repetier-Host #371: Violation of license (and pure asshattery by XYZ)

magnusjjj, 2017-08-26:
Hi. It seems like XYZWare, the printing program that comes with XYZ 3d printers, is just a heavily mutilated repetier host, with anything that looks scary just hidden. Decompiled it, and ran across things like:

"			this.helpToolStripMenuItem.DropDownItems.AddRange(new ToolStripItem[]
			{
				this.repetierHostHomepageToolStripMenuItem,
				this.repetierHostDownloadPageToolStripMenuItem,
				this.manualToolStripMenuItem,
				this.toolStripMenuItem2,
				this.slic3rHomepageToolStripMenuItem,
				this.skeinforgeHomepageToolStripMenuItem,
				this.toolStripSeparator2,
				this.repRapWebsiteToolStripMenuItem,
				this.repRapForumToolStripMenuItem,
				this.thingiverseNewestToolStripMenuItem,
				this.thingiversePopularToolStripMenuItem,
				this.toolStripMenuItem1,
				this.aboutRepetierHostToolStripMenuItem,
				this.donateToolStripMenuItem,
				this.checkForUpdatesToolStripMenuItem,
				this.supportToolStripMenuItem
			});"

Its all just hidden, visibility set to false.
And the standard draconian 'We own everything!' license in there. They also include a whole bunch of other open source libraries, along with statically linking with GPL licensed stuff.
I am sure this isn't news, but..? Couldn't find anything on it.

magnusjjj, 2017-08-26:
[image]

repetier, 2017-08-27:
Wouldn't have guessed that from the look I know. Guess they have taken the latest open source version as base, which is ok so far. Linking GPL stuff makes them GPL but these are not your products, so nothing I would do here. Nonetheless thanks for the information. Always nice to know who is using our software base.