repetier/Repetier-Host #373: Can't compile on Windows 10

pcarew, 2017-09-20:
I'm truing to debug a serial USB connection issue and wanted to be able to modify the code and debug 
I have w10 plus visual studio 2017 C# installed. 
Opentk website is down so I can't pull the binaries 
I tried grabbing opentk source and  building that from scratch, but repetier host still won't compile.  It keeps complaint about opentk not being available 

Any pointers?

repetier, 2017-09-21:
This host version is hopelessly expired. The OpenTK version used for this is even more outdated. We use a quite recent version of OpenTK for our current version from nuget.

The OpenTK sources are available from github and still under development. No idea why they do not repair OpenTK.com. Not working for some month and only source I know for documentation.

pcarew, 2017-09-23:
>"..This host version is hopelessly expired"
So "Repetier Host" should not be used anymore, or are you saying that there is a different version of the source somewhere?

Where do people go now when they want to use 3d printer software on a computer?
How do people directly drive a 3d printer these days?

repetier, 2017-09-23:
https://www.repetier.com
and
https://www.repetier-server.com
is where you get the up to date software with a simple installer.

closed 2017-09-23 by repetier

pcarew, 2017-09-23:
But I can't debug my connection issue with just an executable.
I have repaired a 101hero board. This repaired board works on Pronterface. I have written test code in Java and C# and can confirm that I can open a connection and read data from the board, but in Repetier Host, all I get is: "A Device attached to the system is not functioning" plus the associated popup.
There's no server running, and nothing else is connected to the port.

repetier, 2017-09-23:
Don't know that and searching for the string gives 0 hits. Can you make a screenshot. Normally you see a error message only in the log and a popup asks to enter printer settings to fix the settings.

nophead, 2017-09-23:
"A Device attached to the system is not functioning" is a message from the
OS. Have a look in device manager to see which device it is.

pcarew, 2017-09-23:
Here are some screen shots.
- RepetierHost Connection Problem
- RepetierHost Setup for Port
- Dev Manager showing Port exists
- Win10 Port status
- Pronterface has no issues connecting with the device!

[image]
[image]
[image]
[image]
[image]

repetier, 2017-09-24:
First it is a unusual chipset, so it might be a problem with the driver not supporting some things. The error message comes from the connect part but the catch block covers the complete initialization including.  We also use a derived class from SerialPort to solve some issues with the quite bad c# implementation. Important here the overridden open function:
```
        public new void Open()
        {
            //try
            //{
            bufferSize = base.WriteBufferSize;
            buffer = new byte[base.WriteBufferSize + 1024];
            bufferPos = 0;
            ReadBufferSize = 32768;
            base.Open();
                /*
                ** because of the issue with the FTDI USB serial device,
                ** the call to the stream's finalize is suppressed
                **
                ** it will be un-suppressed in Dispose if the stream
                ** is still good
                */
                GC.SuppressFinalize(BaseStream);
            //}
            //catch
            //{
            //}
        }
```
As you see we resized the buffers here.
After opening we toogle RTS and DTS to true (was set false before open)
```
                Thread.Sleep(10);
                serial.DtrEnable = true;
                serial.RtsEnable = true;
```

Try this in your test connection and maybe you get the same problem then.

pcarew, 2017-09-24:
Thank you, I'll try it and let you know the outcome.
Much appreciated.

pcarew, 2017-09-24:
Thank you,
This really helped. It's the RtsEnable that is causing an issue. 
DTR is fine and works as expected, but RTS is causing the exception.

I am investigating the driver/hardware over this.

pcarew, 2017-09-25:
It is definitely an RTS issue.
I have logged a support question with Silicon Labs about the CP2102N and RTS.
It also looks like if the RTtsEnable call is made *before* the open, everything is OK, but once the port is opened, any call to RtsEnable causes the exception.

DTR works as expected and like many other boards, is used to toggle the reset line.

Here is the RTS exception in detail:
`System.IO.IOException: A device attached to the system is not functioning.

   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   at System.IO.Ports.InternalResources.WinIOError()
   at System.IO.Ports.SerialStream.set_RtsEnable(Boolean value)
   at System.IO.Ports.SerialPort.set_RtsEnable(Boolean value)`

It looks like the .Net SerialStream isn't liking the response from the low level call and thus invoking "WinIOError"

repetier, 2017-09-25:
Ok, good to know. Also it is the first serial driver I know having that problem. It would be no problem to ignore the signal as this is anyway for these printers that reset on RTS and not DTR like most do.
Please let me know if you get a response from them.

pcarew, 2017-09-25:
I should note/add that this is only a C#/.Net issue.
If I do the same operations using Java on Win10, everything is OK!

Any chance that RH could toggle the RTS signal before the open call?
That seems to operate ok on .Net.

repetier, 2017-09-25:
Reset requires a change of pin state, so we also set it false before open already.

Have you tried if repetier-server has the same problem? I'm quite sure it also toggles, but is written in C++. If it works you could use host over server which has also many advantages for print quality and success rate.

pcarew, 2017-09-25:
I haven't used the server. I will look into that in addition

pcarew, 2017-09-25:
Just configured Repetier Server and it connects without issue!

pcarew, 2017-09-25:
It looks like just "Ignoring" the Exception ***isn't*** going to work.
Once the Exception is thrown, the port becomes disabled (possibly by WinIOError) and no further activity is permitted on it.

This means that the best option, if this is **possible,** would be to toggle RTS 'true' before the port is opened.

pcarew, 2017-09-26:
Silicon Labs have confirmed that this looks like a bug in either their device or device driver and have asked for a supporting/demonstrating VisualStudios Project, which I have sent them.

From Silicon Labs:
"..Dear Paul Carew,

Thank you for contacting Silicon Labs Technical Support. This reply is in regards to the case 00152271.

Hi Paul,

That definitely sounds like a bug. Do you mind sharing your VS project with me? That would probably be the quickest route to have me reproduce the issue (and then file a ticket with our driver team), rather than starting a C# project from scratch.

Thanks,
Brian Lampkin
"

repetier, 2017-09-27:
That sounds good. Especially with repetier-server working you can go that road to use host with your printer and in the long run both solutions will work.

pcarew, 2017-09-29:
More from Silicon Labs:
"Hi Paul,

I ran the test on a CP2102N (com 42), and am seeing the exact same error. Thank you for such a good test-case. This will be useful in debugging. I'll keep you updated on my progress

Hi Paul,

I have confirmed that this isn't a configuration issue with the device, so there is some inherent issue with the device's firmware itself, or the driver that is causing the issue. Unfortunately, that is outside my expertise, so I have created an internal ticket with our firmware and driver team to look into this. I will keep you updated with their responses and progress.

Regards,
Brian Lampkin
"