repetier/Repetier-Host #377: injectManualCommand only executes G1 or G0 command once

meyitangp, 2018-01-11:
Hi guys,

First off thank you so much for making RH as accessible as it is. I am writing a custom extension for RH to make it easier to use RH for CNC and Laser operations. 

In my program, whenever I call the IPrinterConnection.injectManualCommand method, I can see that the command is sent to the printer, and I can also see that RH debug view shows the correct command was sent to the printer.
However, the printer only steps once. All other subsequent commands are ignored by the machine.

I compared output of RH's native Manual GUI to mine and it's the same as far as I can tell. What am I missing? I will covert the project to Github soon.

Output of RH native manual GUI works fine every-time I press job button (Note 1 in image below)
`G1 X5 F4800`

Output of my extention executes once and never runs again (Note 2 in image below)
`G1 X5 F4800`

[image]

repetier, 2018-01-12:
You need to be in relative position mode so same command does work every time. See the G91 for case 1 and revert to old mode with G90. In your case I see now G91.

meyitangp, 2018-01-12:
Thank you very much (man that should've been obvious I feel stupid right now :)). I will test on actual machine once I'm back home and report results.

repetier, 2018-01-12:
Also check if it was already in relative mode. Here my move command implementation

```
        private void moveHead(string axis, float amount)
        {
            //RLog.info("move " + axis + " for " + amount.ToString());
            con.connector.GetInjectLock();
            bool wasrel = con.analyzer.relative;
            if (wasrel)
            {
                if (axis.Equals("Z"))
                    con.injectManualCommand("G1 " + axis + amount.ToString(GCode.format) + " F" + con.maxZFeedRate.ToString(GCode.format));
                else
                    con.injectManualCommand("G1 " + axis + amount.ToString(GCode.format) + " F" + con.travelFeedRate.ToString(GCode.format));
            } else
            {
                if (axis == "X")
                    amount += ann.x;
                if (axis == "Y")
                    amount += ann.y;
                if (axis == "Z")
                    amount += ann.z;
                if (axis.Equals("Z"))
                    con.injectManualCommand("G1 " + axis + amount.ToString(GCode.format) + " F" + con.maxZFeedRate.ToString(GCode.format));
                else
                    con.injectManualCommand("G1 " + axis + amount.ToString(GCode.format) + " F" + con.travelFeedRate.ToString(GCode.format));
            }
            con.connector.ReturnInjectLock();
        }
```

That is a new version in 2.0.5 that does not switch between absolute/relative but uses last send positions as reference.

meyitangp, 2018-01-20:
Hi Repetier. Thanks I implemented the old method G91../..G90 toggling due to future ad-dons for the XLNC extension. What was the motivation for you to implement a new algorithm for this case?

closed 2018-01-20 by meyitangp

repetier, 2018-01-20:
Problem was users pressing several times on the button. Then reported position and real position do not match and you could drive it into endstops/bed. Also I got sometimes a relative move in the absolute positioning as it detected that printer was already in relative mode but that was only because the commands were not all send and analysed. That could cause big moves that were not expected. Using last known position as reference and stay in positioning mode solved this problem.