repetier/Repetier-Host #378: Unhandled Exception: Value cannot be Null linux x86_64-2.0.5

rickyrockrat, 2018-01-23:
I repeatedly get this error using either slic3r or cura. Has something to do with printer config, because I can delete the ~/.mono/registry/CurrentUser/software/repetier/printer directory and it starts again. Of course, I loose all my print settings, and it happens every time. Stack dump below. I despise dot net or I would fix it myself.

Unhandled Exception:
System.ArgumentNullException: Value cannot be null.
Parameter name: value
  at Microsoft.Win32.RegistryKey.SetValue (System.String name, System.Object value) [0x00009] in <a07d6bf484a54da2861691df910339b1>:0 
  at (wrapper remoting-invoke-with-check) Microsoft.Win32.RegistryKey:SetValue (string,object)
  at RepetierHost.view.utils.RegMemoryFolder.SetString (System.String r, System.String val) [0x00001] in <e2caed22f11747c098ebda9551e8adde>:0 
  at SlicerSlic3r.Slic3rInstance.StorePrinterSettings () [0x000e0] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at SlicerSlic3r.Slic3rPanel.comboInfillPattern_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x0001b] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at System.Windows.Forms.ComboBox.OnSelectedIndexChanged (System.EventArgs e) [0x00020] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ComboBox.SetSelectedIndex (System.Int32 value, System.Boolean supressAutoScroll) [0x00093] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ComboBox.set_SelectedIndex (System.Int32 value) [0x00000] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ComboBox+ObjectCollection.RemoveAt (System.Int32 index) [0x0004b] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at SlicerSlic3r.Slic3rPanel.UpdateButtons () [0x0001d] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at (wrapper remoting-invoke-with-check) SlicerSlic3r.Slic3rPanel:UpdateButtons ()
  at SlicerSlic3r.Slic3rInstance.ReadPrinterSettings () [0x001a2] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at SlicerSlic3r.Slic3rInstance.SelectPrinter (RepetierHostExtender.interfaces.IPrinter _printer) [0x00023] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at (wrapper delegate-invoke) <Module>:invoke_void_IPrinter (RepetierHostExtender.interfaces.IPrinter)
  at RepetierHost.Main.OnPrinterChanged (RepetierHostExtender.interfaces.IRegMemoryFolder printerKey, System.Boolean printerChanged2) [0x0009d] in <e2caed22f11747c098ebda9551e8adde>:0 
  at RepetierHost.view.FormPrinterSettings.formToCon () [0x001fa] in <e2caed22f11747c098ebda9551e8adde>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.view.FormPrinterSettings:formToCon ()
  at RepetierHost.Main..ctor () [0x01527] in <e2caed22f11747c098ebda9551e8adde>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.Main:.ctor ()
  at RepetierHost.Program.Main (System.String[] args) [0x0001a] in <e2caed22f11747c098ebda9551e8adde>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentNullException: Value cannot be null.
Parameter name: value
  at Microsoft.Win32.RegistryKey.SetValue (System.String name, System.Object value) [0x00009] in <a07d6bf484a54da2861691df910339b1>:0 
  at (wrapper remoting-invoke-with-check) Microsoft.Win32.RegistryKey:SetValue (string,object)
  at RepetierHost.view.utils.RegMemoryFolder.SetString (System.String r, System.String val) [0x00001] in <e2caed22f11747c098ebda9551e8adde>:0 
  at SlicerSlic3r.Slic3rInstance.StorePrinterSettings () [0x000e0] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at SlicerSlic3r.Slic3rPanel.comboInfillPattern_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x0001b] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at System.Windows.Forms.ComboBox.OnSelectedIndexChanged (System.EventArgs e) [0x00020] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ComboBox.SetSelectedIndex (System.Int32 value, System.Boolean supressAutoScroll) [0x00093] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ComboBox.set_SelectedIndex (System.Int32 value) [0x00000] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at System.Windows.Forms.ComboBox+ObjectCollection.RemoveAt (System.Int32 index) [0x0004b] in <c86f0be4f9a847d3b8d63306221e6732>:0 
  at SlicerSlic3r.Slic3rPanel.UpdateButtons () [0x0001d] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at (wrapper remoting-invoke-with-check) SlicerSlic3r.Slic3rPanel:UpdateButtons ()
  at SlicerSlic3r.Slic3rInstance.ReadPrinterSettings () [0x001a2] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at SlicerSlic3r.Slic3rInstance.SelectPrinter (RepetierHostExtender.interfaces.IPrinter _printer) [0x00023] in <abcc901035e5495d8851a86b3656fa7e>:0 
  at (wrapper delegate-invoke) <Module>:invoke_void_IPrinter (RepetierHostExtender.interfaces.IPrinter)
  at RepetierHost.Main.OnPrinterChanged (RepetierHostExtender.interfaces.IRegMemoryFolder printerKey, System.Boolean printerChanged2) [0x0009d] in <e2caed22f11747c098ebda9551e8adde>:0 
  at RepetierHost.view.FormPrinterSettings.formToCon () [0x001fa] in <e2caed22f11747c098ebda9551e8adde>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.view.FormPrinterSettings:formToCon ()
  at RepetierHost.Main..ctor () [0x01527] in <e2caed22f11747c098ebda9551e8adde>:0 
  at (wrapper remoting-invoke-with-check) RepetierHost.Main:.ctor ()
  at RepetierHost.Program.Main (System.String[] args) [0x0001a] in <e2caed22f11747c098ebda9551e8adde>:0

rickyrockrat, 2018-01-23:
The difference between a working printer xml and a broken one is attached.
[printer.patch.txt](https://github.com/repetier/Repetier-Host/files/1654105/printer.patch.txt)

masterbase, 2018-08-28:
I can confirm the error, I can fix it by manually changing the slicer selection in the printer config back to `curraengine`, it crashes as soon as it is set to `sllic3r pursa` or `slic3r`

Non Working:
```
 <value name="slicerSelection"
 type="string">slic3r pursa</value>
```
Working:
```
 <value name="slicerSelection"
 type="string">curaengine</value>
```