repetier/Repetier-Host #379: Issue parsing G-code from Host

meyitangp, 2018-02-23:
Hi guys,

In my RH plugin, I want my plugin to analyse and sometimes modify the g-code that RH is sending to the machine, and also read machine responses.

Currently my idea is to - but this could lead to this issue for large jobs. https://github.com/repetier/Repetier-Host/issues/372:
- Read a g-code file from HDD.
- Read machine state
- Do something with G-code and other processes
- Stream to machine using RH Manual injection method
- Repeat for remaining G-code

I want to maintain as much of RH's streaming mechanisms as possible. How do I:
- When user clicks RH "Print" button, redirect G-code to my plugin for additional processing
- Send back to RH so that RH can stream to machine as usual?

Thanks ahead.

repetier, 2018-02-23:
I do not think this is possible.

You can not see where host is while sending and there is no way to interfere. The only solution is to modify gcode before starting the print. This is normally done by a postprocessor that is called after slicer as external software.

Reading responses is possible. IPrinterConnection has a event for this
event OnResponse eventResponse; // allow others to interpret response

public delegate void OnResponse(string response,ref LogLevel level);

so that allows you to analyse responses. And you can inject commands while printing, just not related to code that will be send.

meyitangp, 2018-02-23:
Oh ok,
When injecting a command during printing, does the Manual Inject method pause current print job before injecting command?

Otherwise alternative is to;
- Lock G-code streaming
- Manually inject command
- Unlock G-code

Will test if RH maintains valid coordinates when this type of operation is performed.

Thank you.

repetier, 2018-02-23:
Injected commands are aded on the fly. e.g. temperature pull is injected every 3 seconds by default. So no pause. Rule is send injected command next if it is there otherwise check job command.

meyitangp, 2018-02-23:
awesome thx.

closed 2018-04-12 by meyitangp