repetier/Repetier-Host #381: "Failed to change owner & group for wx-config…"

thinkyhead, 2018-03-11:
When using the built-in (Sparkle?) **Install Update** option in Repetier-Host the installation fails with the following dialog:

[image]

repetier, 2018-03-11:
Have no real clue why this happens to some users. What helps is manually downloading and installing the files.

thinkyhead, 2018-03-11:
> What helps is manually downloading and installing the files.

Yes, well we are forced to. Good luck on a fix. Maybe it's a bug in Sparkle.

repetier, 2018-03-11:
At least this command is executed by sparkle. So yes, it I think it comes from there. Strangely on my 2 macs I did not get it, but others did as well.