repetier/Repetier-Host #383: python tweakatz script

tecan, 2018-07-10:
include this in the repo if you want.
https://www.thingiverse.com/download:5119674  script from here.

from thing https://www.thingiverse.com/thing:2934923

repetier, 2018-07-10:
What is the python script for?

Also this seems very special, so nothing for a general version. Postprocessors in general are special - either for a special printer or problem. Therefore we include no postprocessors by default.

tecan, 2018-07-11:
there is the RTTTL2GCODE script on that site now too.  along with v2 of tweakatz. also i came into what i think is a bug after i use repetier-host to do a print then use pronterface after i keep getting the line Resend:linenumber status messages from the printer. is there a way to clear it ?

repetier, 2018-07-11:
Ok, I see that you script change stemperature at az=2.24mm. Really special, so not going to add it as a default:-)

Resend is from communication errors when firmware is not happy. This should normally catch up and resend the line and make firmware happy. Except if there is a reason for firmware to never accept a line. In that case, please a log with ack/command enabled where it happens and does not get fixed. Then I can see what is happening and eventually fix it. There is no way to stop it manually except disconnecting as it must be solved be resending.

tecan, 2018-07-12:
https://www.youtube.com/watch?v=JDYsk1KQaHk

tecan, 2018-07-12:
working on a resume print script to strip out the gcode before zposition

tecan, 2018-07-12:
https://github.com/tecan/GCODE-Tools/tree/master/GCODE-Pro

tecan, 2018-07-21:
i think it is more ready now. also i may incorperate a rsa openssl importer/exporter for the gcode files.