repetier/Repetier-Host #386: m117 parsing and triggers.

tecan, 2018-07-28:
m117 parsing strings to launch script. for custom actions like saving z height to a file or playing music during print.

repetier, 2018-07-28:
M117 would be a hack as it is for changing the printer lcd status screen. Hosts already react on many infos returned by firmware, but i guess you want to send host commands to the host. That is not required. If you want a host action just insert
@command
In the gcode and hist will execute the comand. See docs for available commands. Repetier-Server has here some more possibilities and also supports executing external commands  or running scripts from firmware response,

tecan, 2018-07-29:
oh i see it needs a extcommands.xml file

tecan, 2018-07-29:
i tried putting Repetier-Host/database/extcommands.xml in place with proper config. im unsure why it wont run.

tecan, 2018-07-29:
oh all i have to do is @execute fullpathname to binary... i was trying it with gedit and it needs /usr/bin infront

closed 2018-07-29 by tecan

repetier, 2018-07-29:
Extcommands.xml is only required for server a this is more a security risk being a networkable host. After all you should only execute gcodes you generate or trust.